seemoo-lab/openhaystack #112: No location data found

ffischbach, 2022-04-16:
Hey, 
I flashed openhaystack on my esp32. I am working on following setup.

Macbook Version: 12.2.1
Mail Version: 15
Esp32 + OpenHaystack Version 0.5.1

After booting my serial output is the following:
    ⸮⸮ ⸮ ⸮⸮⸮ ⸮   ⸮ ⸮⸮⸮⸮⸮ ⸮⸮ ⸮HFHE⸮⸮⸮0x00,hd_drv:0x00,wp_drv:0x00
    mode:DIO, clock div:2
    load:0x3fff0030,len:6984
    load:0x40078000,len:15184
    load:0x40080400,len:3820
    entry 0x40080698
    [0;32mI (27) boot: ESP-IDF v5.0-dev-2393-gfaed2a44aa 2nd stage bootloader[0m
    [0;32mI (27) boot: compile time 22:40:29[0m
    [0;32mI (27) boot: chip revision: 1[0m
    [0;32mI (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
    [0;32mI (38) boot.esp32: SPI Speed      : 40MHz[0m
    [0;32mI (43) boot.esp32: SPI Mode       : DIO[0m
    [0;32mI (48) boot.esp32: SPI Flash Size : 2MB[0m
    [0;32mI (52) boot: Enabling RNG early entropy source...[0m
    [0;32mI (58) boot: Partition Table:[0m
    [0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
    [0;32mI (68) boot:  0 nvs              WiFi data        01 02 00009000 00005000[0m
    [0;32mI (76) boot:  1 key              unknown          40 00 0000e000 00001000[0m
    [0;32mI (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
    [0;32mI (91) boot:  3 factory          factory app      00 00 00010000 00100000[0m
    [0;32mI (98) boot: End of partition table[0m
    [0;32mI (102) boot_comm: chip revision: 1, min. application chip revision: 0[0m
    [0;32mI (110) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1b03ch (110652) map[0m
    [0;32mI (158) esp_image: segment 1: paddr=0002b064 vaddr=3ffbdb60 size=04668h ( 18024) load[0m
    [0;32mI (166) esp_image: segment 2: paddr=0002f6d4 vaddr=40080000 size=00944h (  2372) load[0m
    [0;32mI (167) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6f3b0h (455600) map[0m
    [0;32mI (337) esp_image: segment 4: paddr=0009f3d8 vaddr=40080944 size=14dc0h ( 85440) load[0m
    [0;32mI (373) esp_image: segment 5: paddr=000b41a0 vaddr=50000000 size=00010h (    16) load[0m
    [0;32mI (384) boot: Loaded app from partition at offset 0x10000[0m
    [0;32mI (384) boot: Disabling RNG early entropy source...[0m
    [0;32mI (395) cpu_start: Pro cpu up.[0m
    [0;32mI (395) cpu_start: Starting app cpu, entry point is 0x4008115c[0m
    [0;32mI (0) cpu_start: App cpu up.[0m
    [0;32mI (412) cpu_start: Pro cpu start user code[0m
    [0;32mI (412) cpu_start: cpu freq: 160000000 Hz[0m
    [0;32mI (412) cpu_start: Application information:[0m
    [0;32mI (417) cpu_start: Project name:     openhaystack[0m
    [0;32mI (422) cpu_start: App version:      v0.5.1-1-gb65a6e6-dirty[0m
    [0;32mI (428) cpu_start: ELF file SHA256:  2c6f8fa6e92ac576...[0m
    [0;32mI (434) cpu_start: ESP-IDF:          v5.0-dev-2393-gfaed2a44aa[0m
    [0;32mI (441) heap_init: Initializing. RAM available for dynamic allocation:[0m
    [0;32mI (448) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
    [0;32mI (454) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
    [0;32mI (460) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
    [0;32mI (466) heap_init: At 3FFC6A68 len 00019598 (101 KiB): DRAM[0m
    [0;32mI (473) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
    [0;32mI (479) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
    [0;32mI (485) heap_init: At 40095704 len 0000A8FC (42 KiB): IRAM[0m
    [0;32mI (493) spi_flash: detected chip: generic[0m
    [0;32mI (496) spi_flash: flash io: dio[0m
    [0;33mW (500) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
    [0;32mI (515) cpu_start: Starting scheduler on PRO CPU.[0m
    [0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
    [0;32mI (546) BTDM_INIT: BT controller compile version [5688ed5][0m
    [0;32mI (546) system_api: Base MAC address is not set[0m
    [0;32mI (546) system_api: read default base MAC address from EFUSE[0m
    [0;32mI (556) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07[0m
    [0;33mW (566) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration[0m
    [0;32mI (1076) open_haystack: using device address: d1 ac a7 ff 63 bd[0m
    [0;32mI (1076) open_haystack: application initialized[0m
    [0;32mI (1126) open_haystack: advertising has started.[0m

(looks fine for me)

But I don't get any location data on Mac Application

[image]

If it's helpful, here's the output from flashing the esp32:

  ./flash_esp32.sh -p /dev/tty.usbserial-0001 "My key"
  esptool.py v3.3
  Serial port /dev/tty.usbserial-0001
  Connecting..........
  Detecting chip type... Unsupported detection protocol, switching and trying again...
  Connecting..............
  Detecting chip type... ESP32
  Chip is ESP32-D0WDQ6 (revision 1)
  Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
  Crystal is 40MHz
  MAC: 78:21:84:77:ee:14
  Uploading stub...
  Running stub...
  Stub running...
  Erasing region (may be slow depending on size)...
  Erase completed successfully in 0.3 seconds.
  Staying in bootloader.
  esptool.py v3.3
  Serial port /dev/tty.usbserial-0001
  WARNING: Pre-connection option "no_reset" was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
  Connecting.......
  Detecting chip type... Unsupported detection protocol, switching and trying again...
  Connecting....
  Detecting chip type... ESP32
  Chip is ESP32-D0WDQ6 (revision 1)
  Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
  Crystal is 40MHz
  MAC: 78:21:84:77:ee:14
  Uploading stub...
  Running stub...
  Stub running...
  Changing baud rate to 921600
  Changed.
  Configuring flash size...
  Flash will be erased from 0x00001000 to 0x00007fff...
  Flash will be erased from 0x00008000 to 0x00008fff...
  Flash will be erased from 0x0000e000 to 0x0000efff...
  Flash will be erased from 0x00010000 to 0x000b4fff...
  Compressed 26080 bytes to 16309...
  Wrote 26080 bytes (16309 compressed) at 0x00001000 in 0.7 seconds (effective 287.9 kbit/s)...
  Hash of data verified.
  Compressed 3072 bytes to 113...
  Wrote 3072 bytes (113 compressed) at 0x00008000 in 0.1 seconds (effective 413.6 kbit/s)...
  Hash of data verified.
  Compressed 32 bytes to 43...
  Wrote 32 bytes (43 compressed) at 0x0000e000 in 0.1 seconds (effective 5.1 kbit/s)...
  Hash of data verified.
  Compressed 672224 bytes to 414299...
  Wrote 672224 bytes (414299 compressed) at 0x00010000 in 6.6 seconds (effective 809.4 kbit/s)...
  Hash of data verified.
  
  Leaving...
  Hard resetting via RTS pin...
  cleanup ...

Sorry if the question is dump...

ffischbach, 2022-04-16:
Also I didn't clicked on the deploy button because I am also working on a Mac with only USB-C ports (mentioned here https://github.com/seemoo-lab/openhaystack/issues/108)

ffischbach, 2022-04-16:
https://github.com/seemoo-lab/openhaystack/issues/104 
Just found this and it does work now... :)

closed 2022-04-16 by ffischbach

liangkei, 2024-12-07:
sorry for bother you.
I found a problem like yours 2 years ago.
I can not connect to my esp32 after https://github.com/seemoo-lab/openhaystack/tree/main/Firmware/ESP32  I do all the steps.
[image]
here's my screenshot
thank you so much

ffischbach, 2024-12-07:
Hey @liangkei, have you tried copying the advertisement key instead?

ffischbach, 2024-12-07:
If I remember correct, my issue was, that I was giving the wrong Key into the flashing command. I did use the symmetric and uncompressed like in the Screenshot below.

The solution was to replace "MY PUBLIC KEY" with the "advertisement key" instead.
`ESP32 % ./flash_esp32.sh -p /dev/cu.usbserial-0001 "MY PUBLIC KEY"`

Hope this helps!

[image]

liangkei, 2024-12-08:
Hello, @ffischbach！
Thanks for your reply!
I replace the public key into the code and reset the device...then it seems works!(because the light from red turn to the green)
but the location looks not accurate now.
Thank you so much! have a good day!