seemoo-lab/openhaystack #115: Device switches from green to yellow and back when turned on/off, but there are no location reports

glatteis, 2022-05-09:
Hi all,

my device does seem to be doing _something_, as it becomes yellow when I turn it off and goes back to green when I turn it back on, but there is no location report. I am completely fine with debugging this, I just need some pointers where to start :)

Best,
Linus

glatteis, 2022-05-09:
Hmm, the simple_server just returns nothing:
`{
	"results": [],
	"statusCode": "200"
}`

glatteis, 2022-05-10:
I've now tried
- a linux laptop
- a raspberry pi
- an ESP32
- 
and nothing. I have no idea what I am doing wrong? I am letting them broadcast the advertisement key (Base64). The reports come back as
```
2022-05-10 10:07:00.772796+0200 OpenHaystack[4502:50857] Headers:
<CFBasicHash 0x60000204ecc0 [0x2095e0000]>{type = mutable dict, count = 8,
entries =>
	0 : Accept = <CFString 0x10553c3c0 [0x2095e0000]>{contents = "application/json"}
	1 : X-Apple-I-TimeZone = CEST
	2 : Content-Type = <CFString 0x10553c3c0 [0x2095e0000]>{contents = "application/json"}
	3 : X-Apple-I-Client-Time = <CFString 0x6000035b8930 [0x2095e0000]>{contents = "2022-05-10T08:07:00Z"}
	4 : X-Apple-I-MD = <CFString 0x60000204ffc0 [0x2095e0000]>{contents = "redacted"}
	5 : X-Apple-I-MD-M = <CFString 0x600001131960 [0x2095e0000]>{contents = "redacted"}
	8 : X-Apple-I-MD-RINFO = redacted
	10 : Authorization = <CFString 0x109e05a90 [0x2095e0000]>{contents = "Basic redacted"}
}
Finished loading the reports. Now decrypt them
Decrypting reports
```

I have three Apple devices around.

glatteis, 2022-05-10:
I used this fix for Apple Mail 16: https://github.com/seemoo-lab/openhaystack/pull/114

Maybe that is the cause? But why would it be?

If someone is willing to test if my devices report correctly, I would like to give them my public/private keypairs so they can test on their end.

glatteis, 2022-05-10:
Okay.. it just worked now. I changed nothing. Huh.

closed 2022-05-10 by glatteis

wtfloris, 2022-05-18:
I think I'm having the same issue, guess I'll just wait for it to start working as well then...

glatteis, 2022-05-18:
I think it just turned out that all of the things I was sending with didn't work correctly. I never got the ESP to work, but for the Raspberry Pi I needed to repeatedly execute the script. I now have an "AliExpress Beacon"-ish board and that works well.

wtfloris, 2022-05-18:
I think the script works because we are getting reports, but the location is "null island", so something is going wrong. 

I don't know if I misunderstand or not but I assume the location data is added by the iPhone that relays the signal, right? So that would be out of reach of the script

glatteis, 2022-05-18:
Oh that's interesting and different from I was encountering. So in the app, you are seeing a dot at lon=0,lat=0?

wtfloris, 2022-05-18:
Not even seeing a dot, but if I click the item, the map centers at lat, lon (0,0)

Can't find anyone else who is having this issue, I guess I'll have to debug and see what the Apple servers are actually returning, might just be a parsing issue

glatteis, 2022-05-18:
Ahhh. That might just be a UI issue from the OpenHaystack app? If you don't see a dot, I assume there are no reports.

glatteis, 2022-05-18:
I'm guessing your device probably doesn't send the BLE advertisements correctly.

If you're so inclined to debug it, I was successful in using the simple_server by the lead dev here:
https://github.com/Sn0wfreezeDev/openhaystack/tree/simple_server
You need to build the "OHServerApp" in Xcode.

This gives you the encrypted reports at a HTTP endpoint, which you can then decrypt for instance using this:
https://github.com/hatomist/openhaystack-python

But this will take a few hours to get it all running. First I would check the logs of the app and/or your device

wtfloris, 2022-05-18:
Thanks for the pointers! I'll let you know what the issue turns out to be :)

glatteis, 2022-05-18:
Let me know if you need any help!

wtfloris, 2022-05-18:
After struggling to get OHServerApp to build (I don't speak Swift and Xcode unfortunately) I've finally got it running haha, but do you know how to use it? I can't seem to find any notes or docs on it

wtfloris, 2022-05-18:
Alright, I'm getting a reply from the servers, but decryption breaks OpenSSL:

```
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "/Users/Floris/openhaystack/openhaystack-python/AirTagCrypto/AirTagCrypto.py", line 69, in decrypt_message
    shared_key = self.__derive_shared_key_from_private_key_and_eph_key(eph_key)
  File "/Users/Floris/openhaystack/openhaystack-python/AirTagCrypto/AirTagCrypto.py", line 32, in __derive_shared_key_from_private_key_and_eph_key
    shared_key = private_key.exchange(ec.ECDH(), public_key)
  File "/usr/local/lib/python3.9/site-packages/cryptography/hazmat/backends/openssl/ec.py", line 162, in exchange
    return _evp_pkey_derive(self._backend, self._evp_pkey, peer_public_key)
  File "/usr/local/lib/python3.9/site-packages/cryptography/hazmat/backends/openssl/utils.py", line 19, in _evp_pkey_derive
    backend.openssl_assert(res == 1)
  File "/usr/local/lib/python3.9/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 212, in openssl_assert
    return binding._openssl_assert(self._lib, ok, errors=errors)
  File "/usr/local/lib/python3.9/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 90, in _openssl_assert
    raise InternalError(
cryptography.exceptions.InternalError: Unknown OpenSSL error. This error is commonly encountered when another library is not cleaning up the OpenSSL error stack. If you are using cryptography with another library that uses OpenSSL try disabling it before reporting a bug. Otherwise please file an issue at https://github.com/pyca/cryptography/issues with information on how to reproduce this. ([_OpenSSLErrorWithText(code=50331798, lib=6, reason=150, reason_text=b'error:03000096:digital envelope routines::operation not supported for this keytype')])
```

"operation not supported for this keytype" seems to be the problem, but no one else seems to have this issue. Am I using the wrong key somewhere? I have noticed that the private keys that AirTagCrypto generates are significantly shorter than my own.

CCxDoomer, 2022-05-25:
Could you please share a runnable server? I am really poor at Swift.

wtfloris, 2022-05-31:
What do you mean "a runnable server"? Does the current version not build for you?