seemoo-lab/openhaystack #122: Release v0.5.2 - no location found - no reports found

Furtivo360, 2022-06-01:
Even after installing the lates openhaystack release v0.5.2 I still experiencing the same issue: 

--> Tag shows green dot as reachable by the application but I cannot see it on the map.

I've tried to change tag's battery and to leave it close to my iphone for a while but still not appearing on map.

mac OS 12.4 (21F79)
Email Version 16.0 (3696.100.31)
OpenHaystack v0.5.2

[image]

Thanks

pawisoon, 2022-06-03:
Same issue here. Everything up to date, green dot next to tag, but still "no location found" :(

Furtivo360, 2022-06-05:
Hi, 

I've been waiting 30mins and more with the application open and the iphone next to the tags...but I still experiencing the same issue...I am not able to see tags on map.

Any help?

Thanks

Furtivo360, 2022-06-06:
Hi, 

So, I've tried to delete the plugin, erase and re-deploy the tag but nothing: "no reports found" still.

What's wrong with my deployment?

Thanks

Furtivo360, 2022-06-07:
Hello!

Any help on this matter? I cannot see any device on the map still :(

Thanks

Furtivo360, 2022-06-12:
Can someone help on this? Anybody with the same issue?

Thanks

Sn0wfreezeDev, 2022-06-21:
Can you compile the project in Xcode and check on the logs? Could be a lot of different things going on here.

Furtivo360, 2022-10-27:
closing this issue as duplicated - following up on [https://github.com/seemoo-lab/openhaystack/issues/148](url)

closed 2022-10-27 by Furtivo360