seemoo-lab/openhaystack #127: Abolish Mail plugin in favor of calling AOSKit::retrieveHeadersForDSID directly

biemster, 2022-06-03:
**Is your feature request related to a problem? Please describe.**
Not really a problem, but I found an easier way of retrieving anisette data that does not require installing a Mail plugin.

**Describe the solution you'd like**
AOSKit has a function `retrieveHeadersForDSID` that can readily retrieve the required anisette data, without needing the entitlements of the Mail program. This is not my work, I just stumbled upon this git repo: https://github.com/vtky/AppleIDAuth (all credits go to @vtky), tested it on Catalina and it works fine with my script: https://github.com/biemster/FindMy

**Describe alternatives you've considered**
Just do nothing, the Mail plugin works fine too. This would only remove an extra step in setting up the whole thing.

**Additional context**
According to Victor who did all the hard work here, the function `retrieveOTPHeadersForDSID` requires a single parameter. Providing `"-2"` as parameter retrieves the same anisette data as the Mail plugin in my tests. The original repo linked above provides a lot of additional info and a compiling XCode project with which I did my tests, and is definitely worth a read (not very long).

biemster, 2022-06-04:
This is actually doable in a couple lines of python too, as a proof of concept:
```python
#!/usr/bin/python

import objc; from Foundation import NSBundle, NSClassFromString

AOSKitBundle = NSBundle.bundleWithPath_('/System/Library/PrivateFrameworks/AOSKit.framework')
objc.loadBundleFunctions(AOSKitBundle, globals(), [("retrieveOTPHeadersForDSID", '')])
util = NSClassFromString('AOSUtilities')

print(util.retrieveOTPHeadersForDSID_("-2"))

```

p1337, 2022-07-16:
Is anyone researching at this topic ?
Are there some people that want make this possible , even maybe a raspi running a script wringing to a DB ?

biemster, 2022-07-16:
There is not much more to research here, you'd still need macOS. Maybe you meant to comment on issue #63 ? There are still people trying out stuff (including me) but there is not much progress atm as far as i can see..

blackknife56, 2022-08-11:
It would be great if we could use a dockerized minimal (e.g. no graphics, just the bare minimum to run the OHserver) version of osx (https://github.com/sickcodes/Docker-OSX) and then run a openhaystack server on it. 
With this setup it may be possible to use even a raspi or a generic linux server.

biemster, 2022-08-11:
I agree, I'm looking into modifying the Catalina recovery BaseSystem.dmg, and running that with https://github.com/foxlet/macOS-Simple-KVM. I'm using a full Catalina with sickcodes/Docker-OSX now, but that clocks in at 30 Gigabytes or so. The BaseSystem I should be able to keep below 1G, and maybe even way below that.
Raspi will be difficult though (and for sure not Catalina), since that is ARM.

Chacha1305Poly20, 2023-10-19:
> @blackknife56 not a MAC expert here but how would you enter apple id credentials without graphical interface?

Seems there has been a long time. There is a project collect the account and password via command-line, so I believe this is possible. https://github.com/icloud-photos-downloader/icloud_photos_downloader

biemster, 2023-10-19:
Also there is pypush now: https://github.com/beeper/pypush