seemoo-lab/openhaystack #128: Esp32C3

thebigboss84, 2022-06-03:
Has anybody was able to successfully build and flash the firmware to esp32C3 chips ? I'm getting error with esp_ble_gap_config_adv_data_raw

bruvv, 2022-08-13:
https://github.com/seemoo-lab/openhaystack/issues/138

olivluca, 2022-08-13:
Did you fix it by installing ESPHome or....?

bruvv, 2022-08-13:
I was already running Esphome so yes :)

thebigboss84, 2022-08-13:
But there is no straight forward fix for the time being

bruvv, 2022-08-13:
Can you try it with the following config?
```
esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: esp-idf
    version: latest
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      # Enable WPA3 support
      CONFIG_WPA_SUITE_B_192: y
      # Specific to Bluetooth usage
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      # Payload size fix - Extend Watchdog timeout
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"
```

DragonHunter274, 2023-01-03:
The sdkconfig needs to be changed and the bootloader offset in the flasher script is wrong for esp32c3.

see here for an example: https://github.com/DragonHunter274/HiP-Badge-openhaystack

Though something is still wrong as the chip gets very hot but it works