seemoo-lab/openhaystack #132: mac os proxy help

Stach-0, 2022-06-10:
hi, does any of you have an ip to the proxy server on mac os?
i need it for the mobile version of app

SinanAkkoyun, 2022-06-27:
I scanned for open ports on my mac and couldn't find anything related to a server running in the background. How can we activate the openhaystack server so that the mobile app will work?

Sn0wfreezeDev, 2022-06-28:
We cannot provide a proxy server for licensing reasons (unfortunately). 
There is an Openhaystack version that runs as a server. 
It's currently only in my private repo and I need to move it to this one

Systm21, 2022-07-03:
I have downloaded and started v0.3.3 from the @Sn0wfreezeDev Repo but i can't get it to work :( 
Is it already in this release, or is it important to compile the code from "simple_server"?  Is it possible to write down a small introduction to get it to work?
Url must be "http://127.0.0.1/getlocationReports" or am i wrong? 
It's such a great project, but i'm not a dev, should be a problem in this case :(

x0r3n, 2022-07-05:
Its working fine.
U should build OHserver and run it, it will listen on localhost:8080 as default in http. U can change it in vapor like             app.http.server.configuration.hostname = "0.0.0.0" and it will listen every where in port 8080 but port u can also change in file AppDelegation in startVapor, but better to use haproxy for it.
After it u can build a mobile app with the server proxy url where listening your OHserver or haproxy

Systm21, 2022-07-06:
@x0r3n ok, it is working right now. At this time, i have no reports, but the server seems to run. 

Is it necessary to run the Openhaystack.app, when using the Server, or is it a standalone Version?

x0r3n, 2022-07-07:
@Systm21  the openhaystack.app dont need to be run, but the Mail app must be run and must be installed the Mail plugin

Systm21, 2022-07-07:
@x0r3n ok, got it. 

There seems to be a Problem with my OHServer.app. In the 0.5.2 Openhaystack.app,  one of the Devices where discovered today (Date and Time is Displayed under the Name), but it is not displayed in the Server ({"error":true,"reason":"Not Found"}). The Problem is, that i can not check the device's needles on the map, because VMWare Esxi doesn't Show the Map (3D acceleration). So i really need to get the Server App to work.

projectsun2, 2022-07-13:
Hi. I have never worked with XCode before. I get this simple_server thing OHServer to run, but there is no response on 8080.
Compiling the normal client works as expected.
What I´m doing wrong?

Astu04, 2022-08-01:
I'm building from source OHServerApp and I'm also having the same issue as the previous comment. 
The console on xcode is empty and if I run it on terminal it's empty too. I even tried getting some log by adding a NSLog("Test") on the do inside the startVapor function on AppDelegate.swift without avail, empty console too. 
I can compile and run the normal openhaystack app without issues.

Systm21, 2022-08-01:
In default configuration, the Server ist listening only on http://localhost:8080/getLocationReports

If you want to have it open in the local Network you have to edit the Vapor Server.
The File is located in the Vapor Package:
Vapor->HTTP->Server->HTTPServer

You can set the IP to 0.0.0.0 an the Port to something different as 8080, if you like. 
0.0.0.0 opens the binding to every IP address, not just localhost.

Astu04, 2022-08-01:
All the tests done were against the same machine with Safari (GET, but I was expecting some error) using either localhost or 127.0.0.1, to that endpoint and to /. I also tried the wildcard 0.0.0.0 but I wasn't successful either. 
I've also tried POST requests with python against the getLocationReports but I get a Err61 Connection refused.

Though in the meanwhile I've been checking the Dadoum/Provision repo and it looks quite promising.

Systm21, 2022-08-01:
You have to POST the json string to the /getLocationReports Server URL.

Astu04, 2022-08-03:
I git cloned the repo, switched to the branch simple-server and I built OHServerApp. I'm using the following python script to test it out: 
`#!/usr/bin/env python3
import requests

url = 'http://localhost:8080/getLocationReports'
data = {'ids': ['xxx']}
r = requests.post(url, json = data)
print(r.text)`
But I'm still getting a 61 connection refused error.

Itheras, 2022-08-03:
You don't need to modify the app itself just use serve Pathto/OHServerApp  -hostname 0.0.0.0 --port 8080 or what ever  port and ip you want.

GarethWright, 2022-08-04:
What's the json request format for http://localhost:8080/getAndDecryptReports?

kaoz3000, 2022-10-12:
I have problems Building the App (Maybe because I a new to XCode)
I downloaded a zip with the simple_server branch and opend the file OpenHaystack.xcodeproj
Xcode Version 14.0.1 gives me this error:

> error project: Failed to resolve dependencies Dependencies could not be resolved because no versions of 'swift-nio-ssl' match the requirement 2.10.4..<3.0.0 and root depends on 'swift-nio-ssl' 2.10.4..<3.0.0.

After closing xcode and opening the same file only this Error are shown:

> /XX/openhaystack-simple_server/OpenHaystack/OpenHaystack.xcodeproj error project: Missing package product 'Crypto'

> /XX/openhaystack-simple_server/OpenHaystack/OpenHaystack.xcodeproj error project: Missing package product 'NIOSSL'

and 3 warning like this: 

> warning build: Run script build phase 'Run swift-format' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase.

After hitting the Built-Button under Products all Apps except OpenHaystack.app and OpenHaystackMail.mailbundle are red.

I now have 10 Issues but none are about the red apps which were not built.
The OpenHaystack App is working.

GarethWright, 2022-10-13:
Open the workspace not the project. That should resolve the issue

kaoz3000, 2022-10-13:
Thank you!
This sounds too simple but, I only have the .xcodeproj file no .xcworkspace
I tried to open different directories but only the OpenHaystack with the .xcodeproj works
[image]
But with same results:
[image]

GarethWright, 2022-10-13:
Out at the moment but I’ll send you my workspace tomorrow. Remind me if I
forget

kaoz3000, 2022-10-15:
@GarethWright A kind reminder :-) I hope then I will be able to compile the iPhone app, too.

@Sn0wfreezeDev why is this essential file (.xcworkspace) missing?

GarethWright, 2022-10-15:
Not essential so much. Just less manual work required to get the ducks in a
row.

kaoz3000, 2022-10-15:
But can you attach the the file? That would help me a lot!

darthnithin, 2022-10-17:
What kind of licensing issues prevent you from hosting a public server?

Itheras, 2022-10-17:
Proxy-in and Apple find my network undocumented api and utilizing it not only without apple blessing but in an unintended way. Then making it public that pushing it too far. So far private hobbyists and projects has been fine let's not push it to no return.

pawisoon, 2022-10-29:
Where I can find the OH Server App? Its not present for me in repo. How to get it? :)

johnbaker26222, 2023-06-25:
Any chance you could post the binary for the server app with 0.0.0.0 and port 8080 please? Thanks!

kerta1n, 2023-09-26:
@Sn0wfreezeDev @GarethWright is there a place where everyone can obtain the .xcworkspace file? Trying to build results in GetReportsBody and ServerReportsFetcher not in scope.