seemoo-lab/openhaystack #138: ESP32 C3 Not booting

bruvv, 2022-07-17:
**Describe the bug**
When I succesfully flash an ESP32 -c3-mini-1 or the ESP32-C3-DevKitC-02 both fail to boot:
```
rst:0x10 (RTCWDT_RTC_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x17a8
load:0x73252029,len:0x2323203a
Invalid image block, can't boot.
ets_main.c 333
SP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install IDF using `./install.sh esp32c3`
3. Build the bootloader with `idf.py build`
4. Flash the ESP32: `./flash_esp32.sh -p /dev/tty.usbserial-14120 "KEY"`
5. Successful flash
6. Use screen to monitor the device: `screen /dev/tty.usbserial-14120 115200`
7. See the error

**Expected behavior**
Not bootloop

**Screenshots**
see error log

**OpenHaystack version:**
0.5.2

**macOS version:**
12.4

olivluca, 2022-07-26:
Same problem here but using the precompiled image

SebastianBar, 2022-08-08:
I've ported OpenHaystack as an ESPHome component and got it to work on both ESP32 & ESP32-C3. If you're into ESPHome, you can check it [here](https://www.barrenechea.cl/projects/openhaystack-esphome/)! 😄 
Hope it helps!

olivluca, 2022-08-08:
Well, it'd help if I wanted to install ESPHome :wink:  (maybe one day but not right now).
However I'm interested in how you're powering the esp32 in your scooter.

bruvv, 2022-08-13:
@SebastianBar you are a hero! Thanks I got it working on a ESP32-C3.

closed 2022-08-13 by bruvv

olivluca, 2022-08-14:
I finally flashed my esp32-c3 with esphome, easier than I thought. Thank you @SebastianBar

olivluca, 2022-08-14:
BTW @SebastianBar the question about how you power the esp32 was a serious one: from the scooter battery with a 5V converter, with a power bank, or...?

SebastianBar, 2022-08-15:
> BTW @SebastianBar the question about how you power the esp32 was a serious one: from the scooter battery with a 5V converter, with a power bank, or...?

Sorry @olivluca, I didn't answer as the question was a little bit out of topic to this post and Alexander & Milan's work over this repo.

<details><summary>Electrical stuff details</summary>
<p>

My scooter is 60v for dual motors, but they have a 12v line that turns on and off according to the scooter's status. Originally that line powers the screen and other scooters' accessories, and I've hooked up a step-down to it into 5v.

On the ESP side, I'm using a Huzzah32 as you can quickly hook up a li-ion battery and keep it alive for "a little bit longer after the scooter is powered down". My pending work with it is to get used to ESP's deep sleep modes, so I could make the Huzzah's battery last a little bit longer, but as it is, it works great (for me).

Note that on any electrical modification, you need to ensure you're not pushing your voltage lines too far. Make sure the supply line CAN provide the amperage your project requires (you may require to explore how the power supply works beforehand). ESP's are pretty low power by themselves tho. As I'm charging a small battery AND powering the ESP, it starts to add up.

Also: I'm not an electrical engineer. Take this with a grain of salt.
</p>
</details>

smierd, 2024-01-24:
@barrenechea I'm getting this error using ESPHome, any ideas? I am not super familiar with platformio and cmake and some of these other tools.

`cc1.exe: error: -march=rv32imc_zicsr_zifencei: unsupported ISA substring '_zicsr_zifencei'`

sdrev, 2024-03-19:
> @barrenechea I'm getting this error using ESPHome, any ideas? I am not super familiar with platformio and cmake and some of these other tools.
> 
> `cc1.exe: error: -march=rv32imc_zicsr_zifencei: unsupported ISA substring '_zicsr_zifencei'`

HI, Did you find any solution? I am having the same error on my ESPHome installed in HA O.S.

Vishal95180, 2025-03-21:
> I've ported OpenHaystack as an ESPHome component and got it to work on both ESP32 & ESP32-C3. If you're into ESPHome, you can check it [here](https://www.barrenechea.cl/projects/openhaystack-esphome/)! 😄 Hope it helps!

Sir, can we port it to esp32 s3 board ?

barrenechea, 2025-03-21:
> HI, Did you find any solution? I am having the same error on my ESPHome installed in HA O.S.

Please open an issue on https://github.com/barrenechea/esphome-config-files with a minimal reproduction example of the issue, as the pasted error does not give me any clue. `seemoo-lab/openhaystack` is not associated with my ESPHome component. This shouldn't be the place to discuss it.

> > I've ported OpenHaystack as an ESPHome component and got it to work on both ESP32 & ESP32-C3. If you're into ESPHome, you can check it [here](https://www.barrenechea.cl/projects/openhaystack-esphome/)! 😄 Hope it helps!
> 
> Sir, can we port it to esp32 s3 board ?

I found no issues compiling it for S3 (although I didn't flash the image to test it E2E, kinda busy right now). The same as the above :)