seemoo-lab/openhaystack #139: OHServerApp

Systm21, 2022-07-22:
Hi everybody, 
i have compiled the OHServerApp with 0.0.0.0 and Port 8099 in Vapor HTTPServer. App runs, but whatever i do, the Server always shows 
`{"error":true,"reason":"Not Found"}`.
Is it possible, that Someone is making the App with 0.0.0.0 IP for me? 

I don't know, what the Problem is, do I have to pay attention to anything when creating the project with xcode? Is it possible to run Openhaystack and OHServer Side by Side, or should the server run alone to avoid any Problems?

Thank You

Itheras, 2022-07-23:
Is working as expected.  a valid ID or the formatting is wrong so the error is correct and exactly what it says. error yes. why?tag not found. how are you requesting the information and what is the full url you are using?

Systm21, 2022-07-23:
Hello, thank you for your reply. 
I used `http://interalip:8099/getLocationReports`. 
Do I also have to define the key in the URL? 
So far, I have assumed that this will be read from the data of the OpenHayStack.app.

Itheras, 2022-07-23:
url looks good a will advise you remove it now. Then there must be a issue with the id how are you fetching it ? Are you using the iOS app? or another method.

Systm21, 2022-07-23:
Is there a method, without using the app? Thought, that the browser can display the undecrypted Report, when i provide the privatekey in the url.

Itheras, 2022-07-23:
You have to send the key id not the private key and the key id must be sent in its base64 form to the server the response then need to be decoded since in wont be in readable form.

Systm21, 2022-07-23:
Yes, i understand. 

But how do i send the key to the Server? Http Requests?

Itheras, 2022-07-23:
Post json array

igormak, 2023-04-30:
Добрый день.

igormak, 2023-04-30:
Good afternoon.

What is the router for: `getAndDecryptReports` and how to use it?

Bei-Ji-Quan, 2023-09-01:
@Systm21,  Hi, could you please tell me how to change the Vapor HTTPServer? I can't find it