seemoo-lab/openhaystack #143: I can't build to ESP32

jaytnw, 2022-08-12:
Hi, When I use "idf.py build" command  in "openhaystack/Firmware/ESP32" I get this error 

```
CMake Error at /Users/xxx/esp/esp-idf/tools/cmake/component.cmake:312 (message):
  Include directory '/Users/xxx/esp/esp-idf/components/lwip/lwip/src/include'
  is not a directory.
Call Stack (most recent call first):
  /Users/xxx/esp/esp-idf/tools/cmake/component.cmake:486 (__component_add_include_dirs)
  /Users/xxx/esp/esp-idf/components/lwip/CMakeLists.txt:149 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "/Users/xxx/Downloads/openhaystack-main/Firmware/ESP32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /Users/xxx/Downloads/openhaystack-main/Firmware/ESP32/build/log/idf_py_stderr_output_38691 and /Users/xxx/Downloads/openhaystack-main/Firmware/ESP32/build/log/idf_py_stdout_output_38691
```

And when I use flash command "./flash_esp32.sh -p /dev/cu.usbserial-0001 "xxxx" I get error 

```
esptool.py v4.2.1
Serial port /dev/cu.usbserial-0001
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 7c:9e:bd:08:da:70
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.2 seconds.
Staying in bootloader.
usage: esptool write_flash [-h] [--erase-all] [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress] [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]] [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '/Users/jay/Downloads/openhaystack-main/Firmware/ESP32/build/bootloader/bootloader.bin'
cleanup ...

```

Help me please, Thank you.

buzzy, 2022-10-02:
How did you generate the public key that you used for "xxxx"?

beckmx, 2023-01-25:
@jaytnw basically the script is telling you that you didnt generate the build files correctly, your esp-idf installation is not fully working, you need to follow all the steps correctly to have esp-idf running

@buzzy that is generated whenever you open the openHaystack app and create a new accesory, a new jey is generated then

closed 2024-08-24 by jaytnw