seemoo-lab/openhaystack #146: Actual AirTag (nRF52832) hardware support for firmware

luke-jr, 2022-08-15:
It would be nice if the OpenHaystack firmware could be loaded onto a real AirTag.

The factory firmware has annoying anti-features (for anti-stalking purposes), requires an iOS device to setup, and only seems to show up on "Find My" once per 15 minute key per nearby iOS repeater.

Itheras, 2022-08-15:
Anti-stalking protection is not disabled in openhaystack and is not dependent on it phones will still notify if a openhaystack programed tag is following you for a period of time. The once per 15 minute or less update interval is not an airtag limitation and more of the normal behavior of the find my application.  The increased amount of updates in openhaystack is due on how the application works compared to the find my app not the tags themselves.  airtags are currently not supported in this app because of the handling of the private and rotating keys so no increased updates that way at this moment.  and flashing an airtag will prevent it from working with the find my app will remove the proximity finding. if you want less anti-stalking disable the speaker in a regular tag and you won't sacrifice find my functionality that's one of the main differences. if you want more updates I recommend just buying a cheap airtag sized nrf51 or 52 tag. Anything I can help let me know.

luke-jr, 2022-08-16:
>Anti-stalking protection is not disabled in openhaystack and is not dependent on it 

By this, I mainly mean the annoying sounds they make when you pick them up each day.

>phones will still notify if a openhaystack programed tag is following you for a period of time.

Probably not if the key gets changed often enough?

>The once per 15 minute or less update interval is not an airtag limitation and more of the normal behavior of the find my application. 

If a new key is used every minute, iOS devices don't know it's the same one, so they can't implement relay-side anti-stalking, and don't know to throttle reports. Admittedly, this may require additional changes to OpenHaystack unrelated to adding hardware support (this issue).

>flashing an airtag will prevent it from working with the find my app will remove the proximity finding

I have no interest in the Find My app. I only use it due to lack of a better alternative (it runs in a VM and a cronjob uploads its cached coordinates to my own server).

> if you want less anti-stalking disable the speaker in a regular tag 

Then the explicit "make a sound" feature won't work either ;)

(Yes, that may require more OH changes to add too...)

>if you want more updates I recommend just buying a cheap airtag sized nrf51 or 52 tag.

Does such a thing exist w/ a battery in a child-safe form factor?

Itheras, 2022-08-16:
What do you think of this one. I am working on the firmware at this very moment.

https://a.aliexpress.com/_mrHmPbA

luke-jr, 2022-08-16:
By child safe, I mainly mean it will survive the child spilling on it - not a raw PCB like I thought was required by OH.

Currently I clip the airtag inside my toddlers' clothes (using [this holder](https://www.amazon.com/gp/product/B093BZGWS7)).

Browsing Aliexpress more, I found [this beacon with a built-in clip](https://www.aliexpress.com/item/3256802430820638.html). Ironically, it is also a nRF52832...

Itheras, 2022-08-16:
That one is nice . If I can recommend something use nrf51 . They are cheap an easy to program with a simple and cheap stink. Nrf52 requires a more expensive jlink.

Check [this one](https://a.aliexpress.com/_mPRVJe8)

luke-jr, 2022-08-16:
I see. Maybe that is an option.

A reviewer noted "32.768KHz oscillator pad was unpopulated. which means you have to use the internal low frequency clock which is supposed to be less power efficient. The rest of the board was fine."

Would this matter for OpenHaystack?

What are the differences between Beacon vs RSSI?

Itheras, 2022-08-16:
Power difference is slightly higher but the difference is negligible with or without the extra oscillator battery will last more than 3 years with the correct firmware. Beacon and RSSI is not important they are the same devices that just the factory installed firmware but since they will be reflashed it won't matter just make sure if you get it is the version with the case.

luke-jr, 2022-08-16:
Actually, I just noticed my childrens' smart watches (and the PineTime) have a nRF52832 chipset too

https://www.aliexpress.com/item/2255800371394199.html

luke-jr, 2022-10-01:
Is there a trick to searching Aliexpress? Looking for one with a button and speaker too.

[These](https://www.aliexpress.us/item/3256804443889679.html) seem nice, but don't say what they have inside :/