seemoo-lab/openhaystack #147: stuck at listening for gbd

hikaru78, 2022-08-29:
**Describe the bug**
Generic nrf51822 beacon
issues flashing with openocd

Log

paulnguyen@Pauls-MBP paulnguyen % openocd -f share/openocd/scripts/interface/stlink-v2.cfg -f share/openocd/scripts/target/nrf51.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.292761
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for nrf51.cpu on 3333
Info : Listening on port 3333 for gdb connections

have tried multiple tags and multiple stlinks

Itheras, 2022-08-29:
openocd -f share/openocd/scripts/interface/stlink.cfg -f share/openocd/scripts/target/nrf51.cfg
telnet localhost 4444

hikaru78, 2022-08-31:
was a stupid error. did not realise i had to open a second terminal window to telnet into gbd

closed 2022-09-02 by hikaru78