seemoo-lab/openhaystack #148: location not found with green dots

DAZABAMUKER, 2022-09-07:
I did a lot from issue but still It shows location not found with green dots 
I have iPhones besides me...
I Xcode shows these reports..

2022-09-07 15:17:18.977433+0900 OpenHaystack[57957:2676292] Calling IOPPFGetProperty thermally_optimized_maps!
2022-09-07 15:17:18.977475+0900 OpenHaystack[57957:2676292] Cannot find plugin in ioreg!
2022-09-07 15:17:18.981706+0900 OpenHaystack[57957:2676292] Metal API Validation Enabled
2022-09-07 15:17:19.153880+0900 OpenHaystack[57957:2676292] [core] "Error returned from daemon: Error Domain=com.apple.accounts Code=9 "(null)""
2022-09-07 15:17:19.365123+0900 OpenHaystack[57957:2676733] Requesting data for (
    "4vHF9SiJoAYcxLjJTtSpO5/zcGBsgdDSxh/l+aWQUAo="
)
2022-09-07 15:17:19.365329+0900 OpenHaystack[57957:2676733] Query : {
    search =     (
                {
            endDate = 1662531439365;
            ids =             (
                "..."
            );
            startDate = 1660717039365;
        }
    );
}
2022-09-07 15:17:19.380565+0900 OpenHaystack[57957:2676292] Headers:
<CFBasicHash 0x60000361f840 [0x1f51f4000]>{type = mutable dict, count = 8,
entries =>
	0 : Accept = <CFString 0x10530cab8 [0x1f51f4000]>{contents = "application/json"}
	1 : X-Apple-I-TimeZone = GMT+9
	2 : Content-Type = <CFString 0x10530cab8 [0x1f51f4000]>{contents = "application/json"}
	3 : X-Apple-I-Client-Time = <CFString 0x600002cce010 [0x1f51f4000]>{contents = "2022-09-07T06:17:19Z"}
	4 : X-Apple-I-MD = <CFString 0x60000361e800 [0x1f51f4000]>{contents = "AAAABQAAABD2s2reu8NPnt9LcHF6/GXgAAAAAQ=="}
	5 : X-Apple-I-MD-M = <CFString 0x6000008451f0 [0x1f51f4000]>{contents = "pvFpH35nr99bIRnP7BFF9WGBsWHFrCwbcnTzKiNr1PUDjopBKjymlLIzcjIqJ001ucl11PVtsm+"}
	8 : X-Apple-I-MD-RINFO = 50660608
	10 : Authorization = <CFString 0x127e04ad0 [0x1f51f4000]>{contents = "Basic MTAxNTYyOTc5N UFBTkFBQUFCTHdJQUF FHTVlGVklSRG1kekxtbGpiRzkxWkM1aGRYUm92UUNySnNrMWljSnk4Qm1oajdvanAzTzRvS1hvQ2lsZW5NS3piZE9KUDNPZ2xILU9kZG9iZEhTVUtoU2pvSDc3ZUhNWE5Uh2aG5icVU1TmhsN0hoR1JJU3RnSDllMDUzOUp2T3JZNFZ1dXJQa0wzZEQtOEJsZXZ6Zm5CV1FiS0dDc0kwd04xSko5MmJYZG9icG1RUEZejNEd35+"}
}
Finished loading the reports. Now decrypt them
Decrypting reports
2022-09-07 15:17:19.955733+0900 OpenHaystack[57957:2677320] [IconManager] No config pack found for key Central Jpn Custom Icons
2022-09-07 15:17:41.623506+0900 OpenHaystack[57957:2676292] [logging-persist] cannot open file at line 45530 of [9ff244ce07]
2022-09-07 15:17:41.623578+0900 OpenHaystack[57957:2676292] [logging-persist] os_unix.c:45530: (0) open(/var/db/DetachedSignatures) - Undefined error: 0

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**OpenHaystack version:**
0.5.2

**macOS version:**
`12.5.1`

**Additional context**
Add any other context about the problem here.

Furtivo360, 2022-10-27:
Same issue here.

Any solution/help?

Thanks

shadowHacker2020, 2022-12-10:
same issue here ,  any workaraound ?

emreisik95, 2023-06-26:
Same here. I think this repo is not up to date