seemoo-lab/openhaystack #150: uploading to esp8266 fails

Arthur-Emmanuel45, 2022-09-11:
can someone help me with this error when trying to run a simple blink code to esp8266 using platformio in vs code

Processing esp01_1m (platform: espressif8266; board: esp01_1m; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 (4.0.1) > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30100.210531 (3.1.0)
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 1.200.0 (2.0)
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- thinger.io @ 2.25.2
|-- DHT sensor library @ 1.4.4
|   |-- Adafruit Unified Sensor @ 1.1.6
|-- LiquidCrystal_I2C @ 1.1.4
|   |-- Wire @ 1.0
|-- LiquidCrystal @ 1.0.7
|-- Adafruit Unified Sensor @ 1.1.6
|-- Wire @ 1.0
Building in release mode
Compiling .pio\build\esp01_1m\src\main.cpp.o
Generating LD script .pio\build\esp01_1m\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\esp01_1m\lib4da\thinger.io\ThingerClient.cpp.o
Compiling .pio\build\esp01_1m\lib4da\thinger.io\util\md5.cpp.o
Compiling .pio\build\esp01_1m\libba0\Adafruit Unified Sensor\Adafruit_Sensor.cpp.o
Archiving .pio\build\esp01_1m\lib4da\libthinger.io
Archiving .pio\build\esp01_1m\libba0\libAdafruit Unified Sensor.a
Compiling .pio\build\esp01_1m\lib8f4\DHT sensor library\DHT.cpp.o
Compiling .pio\build\esp01_1m\lib8f4\DHT sensor library\DHT_U.cpp.o
Compiling .pio\build\esp01_1m\lib4b8\Wire\Wire.cpp.o
Archiving .pio\build\esp01_1m\lib8f4\libDHT sensor library.a
Compiling .pio\build\esp01_1m\libc8f\LiquidCrystal_I2C\LiquidCrystal_I2C.cpp.o
Archiving .pio\build\esp01_1m\lib4b8\libWire.a
Archiving .pio\build\esp01_1m\libc8f\libLiquidCrystal_I2C.a
Compiling .pio\build\esp01_1m\lib7bc\LiquidCrystal\LiquidCrystal.cpp.o
Archiving .pio\build\esp01_1m\libFrameworkArduinoVariant.a
Compiling .pio\build\esp01_1m\FrameworkArduino\Crypto.cpp.o
Archiving .pio\build\esp01_1m\lib7bc\libLiquidCrystal.a
Compiling .pio\build\esp01_1m\FrameworkArduino\Esp-frag.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\Esp-version.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\FS.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\FSnoop.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\LwipDhcpServer-NonOS.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\LwipDhcpServer.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\LwipIntf.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\LwipIntfCB.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\Schedule.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\StackThunk.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\StreamSend.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\TypeConversion.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\Updater.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp01_1m\FrameworkArduino\abi.cpp.o
[... 85 lines cut ...]
Compressed 265040 bytes to 195054...
Writing at 0x00000000... (8 %)
Traceback (most recent call last):
  File "C:\Users\DHAKA\.platformio\packages\tool-esptoolpy\esptool.py", line 4582, in <module>
    _main()
  File "C:\Users\DHAKA\.platformio\packages\tool-esptoolpy\esptool.py", line 4575, in _main
    main()
  File "C:\Users\DHAKA\.platformio\packages\tool-esptoolpy\esptool.py", line 4122, in main
    operation_func(esp, args)
  File "C:\Users\DHAKA\.platformio\packages\tool-esptoolpy\esptool.py", line 3424, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\DHAKA\.platformio\packages\tool-esptoolpy\esptool.py", line 149, in inner
    return func(*args, **kwargs)
  File "C:\Users\DHAKA\.platformio\packages\tool-esptoolpy\esptool.py", line 874, in flash_defl_block
    self.check_command("write compressed data to flash after seq %d" % seq,
  File "C:\Users\DHAKA\.platformio\packages\tool-esptoolpy\esptool.py", line 462, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\DHAKA\.platformio\packages\tool-esptoolpy\esptool.py", line 425, in command
    self.write(pkt)
  File "C:\Users\DHAKA\.platformio\packages\tool-esptoolpy\esptool.py", line 388, in write
    self._port.write(buf)
  File "C:\Users\DHAKA\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout
*** [upload] Error 1

Filo301, 2022-09-12:
Looks like bad platform config or something. Anyway, OHS is for bluetooth devices only

obitvn, 2022-12-07:
The firmware only supports ESP32, not esp8266. Esp8266 doesn't have Bluetooth