seemoo-lab/openhaystack #152: Android beacon app

luke-jr, 2022-09-13:
It would be nice if there was a simple app to broadcast beacons from an Android device (eg, in place of a microcontroller).

biemster, 2022-09-15:
To the best of my knowledge it requires root rights to change the Bluetooth MAC on android, please correct me if I'm wrong because I would like this too.

luke-jr, 2022-09-15:
I don't know. root would be acceptable for many use cases (just think of all the 3G-only phones that are no longer useful...).

biemster, 2022-10-28:
For me the first bytes of the pubkey are shown as XX:XX:XX in the cloned advertisement, is there an option I'm missing?

luke-jr, 2022-10-29:
>You can simply do this from android nRF app, set MAC header, and payload according to /firmware and pubkey.

I mean an app that does it automatically, without a user constantly involved...

>Just make your tag/phone believe it's stolen (see above),

You didn't explain above. And by "stolen", I assume you mean away from their iOS device? Shouldn't need to actually be lost/stolen...

luke-jr, 2022-10-29:
>Broadcasting constantly from android is not reliable os will kill it eventually.

Plenty of apps broadcast constantly without issues.

>It's considers itself stolen/away when disconnected from BLE/wifi for ~30mins. (e.g. left in the park)

Away, yes (but just BLE - airtags don't support wifi...). Stolen/lost can only be manually triggered, and prevents it from being re-registered to another user.

Hitmanreis, 2024-03-17:
Up

jcardus, 2025-09-01:
I tried adveritising the same data the firmware does, but for some reason it doesn't work.

biemster, 2025-09-13:
You can indeed change the address on rooted devices, so it should be possible to create a solution for such devices.
The google FMDN works also over BLE, with a keystone advertisement. It has been reversed already as well, especially here: https://github.com/leonboe1/GoogleFindMyTools/

biemster, 2025-09-13:
It should be straightforward to advertise the FMDN keystone on a googleless android, you could have a look at https://github.com/biemster/FindMy/blob/main/ESP32/findmy_google.py how such a frame should be constructed. The requirement for MacOS is already for really long not there anymore, people moved to using https://github.com/biemster/FindMy/blob/main/pypush_gsa_icloud.py when the pypush developer realized they are getting the search-party-token in the login responses.