seemoo-lab/openhaystack #154: Idea: Some way for Android phones to participate in "Find My" beacon uploads

luke-jr, 2022-09-19:
Very few Apple devices in my area. It would be nice if I could opt-in my Android device(s) to submit "Find My" beacons to Apple's network.

Bonus feature ideas:
- an option to submit only my own, or any/all seen
- geofencing to only submit when I'm not at/near home
- perhaps make GPS coordinates "fuzzy" so as to avoid too precise positioning of my phone

luke-jr, 2022-09-20:
(Actually, for the "only submit my own" mode, it could just submit the beacons directly to a non-Apple server...)

biemster, 2022-10-29:
To my understanding reporting beacons need
1) A signed in iDevice. This is also required for obtaining one's own reports, which is not yet reversed enough to do on non-Apple hardware (because of Anisette data and search-party-token)
2) A signing key from the hardware key store in iDevices, which is connected to the signed in session. To my best knowledge there is no active research how to generate valid ones of those.

There is a bit of research into 1) in for example AltStore (and #63), and 2) is not useful without having full 1), so there seems to be a long road ahead for this.