seemoo-lab/openhaystack #155: Unclear instructions

buzzy, 2022-10-01:
I am trying to get OpenHaystack on my ESP32. The instructions do not explain if I need to actually flash the firmware to the ESP32 first or if the "deploy" button in the GUI does this for me?

In either case, I have compiled by own firmware with ESP-IDF and ready to flash it to my ESP32, but how exactly do I generate this "base64 of my public key"? "we generate a public-private key pair on an elliptic curve (P-224)".. ok, how?

Seems I have to do this manual flashing first, because clicking on "deploy" and picking ESP32 (and port) gives me the error "deploying to microbit (?) failed". I assume it's because it wants a pre-flashed ESP32 with a special "key"-partition to read/write to/from?

Please advice. Thanks.

iLazlow, 2022-10-25:
There seems to be a bug in the build in flashing process. It failed every time for me because it was trying to flash a microbit. To get your base64-Key-Pair you have to add a new device in openhaystack. Then right click copy your advertisement base64 key and flash your ESP with the ESP-IDF. After that you have to mark your ESP as deployed in openhaystack (right click menu).

TomOMaley, 2023-03-13:
I have also questions regarding the instructions. Can I flash esp32 directly via the App using the deploy button? My Chip is  ESP32-D0WDQ6 (revision v1.0). When I try to deploy I get the message "Could not deploy - deploying to microbit fails. ... " Does that mean I have to flash the esp32 beforehand, and if so, how? The instruction here (https://github.com/seemoo-lab/openhaystack/tree/main/Firmware/ESP32) says that I have to implement ESP-IDF. I was able to do it but cannot run "idf.py build". I assume that ESP-IDF is not on the right $PATH ??? What exactly means: "With the ESP-IDF on your $PATH, you can use idf.py to build the application from within this directory: ... " What is the right $PATH???
I sincerely hope there is someone out there who can help me. I've been sitting on this problem for days.

buzzy, 2023-03-13:
> I have also questions regarding the instructions. Can I flash esp32 directly via the App using the deploy button? My Chip is ESP32-D0WDQ6 (revision v1.0). When I try to deploy I get the message [...]

Just means you need to put idf.py somewhere in you $PATH. Either add your path to $PATH or simply put it in like /usr/bin

TomOMaley, 2023-03-13:
Thank you. but I have no clou what $PATH means?

buzzy, 2023-03-13:
> Thank you. but I have no clou what $PATH means? Did you mean ESP-IDF or idf.py? Thought the latter is a command?

"What is $PATH?" on Google gives you the answer...

TomOMaley, 2023-03-20:
Thank you @buzzy. Got it. But still no succes. The build/openhaystack.bin is not found when I run idf.py build. I see it in the packagemanager of the OpenHaystack App. Does that mean that the whole App must be on the $PATH?

buzzy, 2023-03-20:
What do you mean "not found"? the "idf.py build" will CREATE the .bin file.