seemoo-lab/openhaystack #156: Mail crash when is opened (macOS 12.6 mail 16.0)

MoutardeForte, 2022-10-19:
Hello, Mail crash when is opened

Steps to reproduce the behavior:
1. Check the plugin box in mail
2. retart mail
3. crash

Screenshot
[image]

OpenHaystack version : 0.5.2 (1)

macOs version : 12.6

MoutardeForte, 2022-10-20:
Hello I have tried with this script : 
https://gist.github.com/noahwilliamsson/08500bb6417015b7d6023e0b42cac34e#file-openhaystack-fix-compat-sh

Here the return : 
`macOS version (major, minor):       12.6
Mail.app plugin compatibility UUID: A4B49485-0377-4FAB-8D8E-E3B8018CFC21

Mai bundle claims to support the following plugin compatibility UUIDs:
Array {
    25288CEF-7D9B-49A8-BE6B-E41DA6277CF3
    224E7F96-2099-499C-A501-63FB68C79CD2
    D985F0E4-3BBC-4B95-BBA1-12056AC4A531
    6FF8B077-81FA-45A4-BD57-17CDE79F13A5
    A4B49485-0377-4FAB-8D8E-E3B8018CFC21
}

>>> OK, /Users/lubin/Library/Mail/Bundles/OpenHaystackMail.mailbundle/Contents/Info.plist already claims to be compatible with A4B49485-0377-4FAB-8D8E-E3B8018CFC21

>>> Using  sudo spctl  to whitelist the mailbundle in GateKeeper (you might be asked to enter your login password)...

+ sudo spctl --add --label 'Allow OpenHaystack Mail.app plugin to load' /Users/lubin/Library/Mail/Bundles/OpenHaystackMail.mailbundle
+ set +x

>>> Finally, open Mail.app and goto Preferences > General > Manage plugins and check the OpenHaystackMail.mailbundle plugin.`

But always same problem mail crash when opened....

Inf3rnus187, 2022-10-23:
Hi.

I was having same troubles. MacOS 12.6 and mail V16. 

First in Xcode I have changed the destination version from macOS 11 to 12.3 for open haystack et mail bundle before build the app.  
Same issue so after searching, the trick for me was to connect my Mac to iCloud in settings.

MoutardeForte, 2022-10-25:
Thanks a lot ! I log my business account into icloud and it work now.