seemoo-lab/openhaystack #157: Add suport for plugin for MacOS 13.0 Ventura and Mail app Version 16.0 (3731.200.110.1.12)

pawisoon, 2022-10-25:
Hi there,
Can we add support for MacOS 13.0 Ventura and Mail App app Version 16.0 (3731.200.110.1.12)? 

I'm getting plugin error after update. 
[image]

pawisoon, 2022-10-26:
@jordan754 no need. I created my fork and made pull request adding UUID support for MacOS 13. Waitig for it to be merged by owner.

pawisoon, 2022-10-26:
Btw, this IG account looks like total fake and stolen pics. Please be aware whoever sees this thread in the future.

jrusi, 2022-10-27:
I've tried to load the file manually but still get the incompatibility error
1. I close the Openhaystack and Mail app
2. Remove mailbundle: rm -r /Users/user/Library/Mail/Bundles/OpenHaystackMail.mailbundle
3.  sudo spctl --master-disable  
4. Replace the Info.plist with the one created by pawisoon in: /Users/user/Applications/OpenHaystack.app/Contents/Resources/OpenHaystackMail.mailbundle/Contents
5. sudo spctl --master-enable
6. Run openhaystack app and install plugin
7. Run Mail app and enable the plugin
8. After restart the plugin is being disabled with the error message shown in post above

Furtivo360, 2022-10-27:
Hi, I've been able to update the Info.plist and to re-enable the plugin!

OpenHaystack is now opening once again, thanks! 👍 

But I still having the same issue with the "No report found"

I can see the green dots next to my accesories but no location on the map...

Any help on this boyz and gyrlz? Please 

[image]

jrusi, 2022-10-27:
Do you have the green dot in the top right corner?
[image]

Furtivo360, 2022-10-27:
Yep, green dot there too 

[image]

Katuma2345, 2022-10-28:
I'm sorry if I sound like a beginner (because I am 😂) but can someone please run me through how to do all of this for MacOS Ventura?  I've got the same compatibility error (mail 16.0). So far all I have done is open the file in Xcode because it wouldn't let me install it (can't check for malicious software or something like that error message). If I'm doing it all wrong please do let me know, this is my first time doing anything close to coding in years, and certainly the first time using an apple product. So I'm sorry for bothering everyone but a literal step-by-step instruction process would be extremely helpful for someone like me and I would really appreciate it :)

Katuma2345, 2022-10-28:
I've got the same problem, it says no reports found :(

parnz, 2022-10-29:
> @jordan754

Can we add the UUID in advance to support the future release of macOS?
or wildcard can be used?

Ex: 
Supported13.*PluginCompatibilityUUIDs
or 
Supported*.*PluginCompatibilityUUIDs

pawisoon, 2022-10-29:
@parnz  just use my fork or add changes I made in PR https://github.com/seemoo-lab/openhaystack/pull/158

pawisoon, 2022-10-29:
Can owner of this repo już allow PR?  #158

exrector, 2022-10-30:
Same error. Need an update to version 16
[image]

idockplatform, 2022-10-31:
I tried also to edit the Info.plist, but I get still this error:

[image]

Furtivo360, 2022-10-31:
> How can I fix this problem? I tried also to edit the Info.plist, but I get still this error:
> 
> [image]

Have you ad-hoc signed the file?

`codesign --force --deep -s - OpenHaystackMail.mailbundle`

idockplatform, 2022-10-31:
Yes. :(

pawisoon, 2022-11-01:
@idockplatform what helped for me was to remove the original plugin from folder. Cleaning build and re-running it again. Then I added plugin again(a fresh one this time with new .plist UUIDs) and enabled it succesfully on MacOS 13 and Mail 16.0

Furtivo360, 2022-11-02:
> Yep, green dot there too
> 
> [image]

Any help on this? Anybody solved it?
Thanks

tmayberry, 2022-12-01:
I had the same problem, still saying it was incompatible after changing the Info.plist to include the new compatibility string.  The problem turned out to be that the default setting for the project in Xcode is to compile to Rosetta (the compatibility layer for new M1/M2 Macs) and this is not allowed for Mail plugins.  Not sure if this just started with Mac OS 13 or what, because it was working for me before, but you can fix it by going to Product -> Destination in Xcode and making sure it is set to My Mac instead of My Mac (Rosetta).

bruvv, 2022-12-09:
@tmayberry can you eleborate a bit more where we can find `product` ?

tmayberry, 2022-12-09:
> @tmayberry can you eleborate a bit more where we can find `product` ? Is this in Xcode program itself?

Yes, you have to recompile the plugin from Xcode.

bruvv, 2022-12-09:
As you already did it, can you add the files to your comment :) ?

DanielBustillos, 2022-12-24:
> @idockplatform I had the same problem, still saying it was incompatible after changing the Info.plist to include the new compatibility string. The problem turned out to be that the default setting for the project in Xcode is to compile to Rosetta (the compatibility layer for new M1/M2 Macs) and this is not allowed for Mail plugins. Not sure if this just started with Mac OS 13 or what, because it was working for me before, but you can fix it by going to Product -> Destination in Xcode and making sure it is set to My Mac instead of My Mac (Rosetta).

This worked for me. Using macOS 13.1. I also used the file of @pawisoon