seemoo-lab/openhaystack #162: Add support for plugin for macOS 13.1 Ventura and Mail app Version 16.0 (3731.300.101.1.3)

jrusi, 2022-12-15:
After 13.1 upgrade there is an issue with the plugin. It get's rejected and disabled in Mail.app

I've checked the UUID and it seems that it is the same as the previous one:
<string>890E3F5B-9490-4828-8F3F-B6561E513FCC</string>

But when I add new entry in /Applications/OpenHaystack.app/Contents/Resources/OpenHaystackMail.mailbundle/Contents/Info.plist:
"Supported13.1PluginCompatibilityUUIDs" = ( "890E3F5B-9490-4828-8F3F-B6561E513FCC" );
and enable the plugin, Mail app gives me this error:
The item "OpenHaystackMail.mailbundle" is corrupted and cannot be opened. Move this item to the Trash.

Any ideas how to fix it?

Myzel394, 2022-12-16:
Hey @jrusi I'm also trying to get this to work. Did you find a workaround?

DanielBustillos, 2022-12-24:
I have the same issue. I used the file of #157 and update the macOS 13.1 mail 16.0 UUIDD using the one in 

`/Applications/Mail.app/Contents/Info PluginCompatibilityUUID `

But I keep getting the error line

> Contact the makers of these plug-ins for versions that are compatible with Mail 16.0.

zwells, 2022-12-28:
What I did that got it working again:

Turn off mail and openhaystack
Remove: ~/Library/Mail/Bundles/OpenHaystackMail.mailbundle and ~/Library/Application Support/Mail/Plug-ins/*
Edit: /Applications/OpenHaystack.app/Contents/Resources/OpenHaystackMail.mailbundle/Contents/info.plist 
Add this entry:
[image]
Turn on openhaystack, it will ask to install the plugin, install it in: /Users/username/Library/Mail/Bundles/
it will open mail, close it
Disable gatekeeper `sudo spctl --master-disable`
Open mail and enable plugin
Allow access prompt
Mail will ask to restart
Enable gatekeeper `sudo spctl --master-enable`

DanielBustillos, 2022-12-28:
What worked for me is compiling the binaries with Xcode selecting my Mac as a target and adding manually the entries commented by @zwells .

danielsippel, 2023-01-31:
@DanielBustillos Can you provide a compiled version with the entries for 13.1 & 13.2 ?
Or some lines of how to compile?

Actually I can't see if my boat is still where it should be :-)

mishailovic, 2023-02-15:
Hey guys, little update on this one. I bumped versions of mailplugin and built it using GitHub actions. You can grub it [Here](https://github.com/mishailovic/openhaystack/releases/download/v0.5.3/OpenHaystack.zip.zip) 
(Download and unpack .zip file and go to the Contents/Resources you will find mail plugin here.)
Install plugin and use latest version of openhaystack from [here](https://github.com/seemoo-lab/openhaystack/releases) 
Works for me on latest macos (13.2.1)

pcace, 2023-02-16:
Hi, i just tried your version. Since i have never ran openHaystack successfully i dont know how i would know if it worked. would there be anything shown in mail/extensions? with your version there is nothing shown here for me (M1 Mac, 13.1)
How do i know if it worked?

mishailovic, 2023-02-16:
@pcace yeah, this little dot should become green. I think apple silicon is another pain, I'm not really sure what to do with it

Furtivo360, 2023-02-16:
Hi! This new version v0.5.3 keeps crashing.
Cannot open it.

Any thoughts?

Thanks

mishailovic, 2023-02-16:
> Hi! This new version v0.5.3 keeps crashing. Cannot open it.
> 
> Any thoughts?
> 
> Thanks

Yes, this is expected. Install mail plugin from 0.5.3 and use 0.5.2 from core repository

Furtivo360, 2023-02-16:
> > Hi! This new version v0.5.3 keeps crashing. Cannot open it.
> > Any thoughts?
> > Thanks
> 
> Yes, this is expected. Install mail plugin from 0.5.3 and use 0.5.2 from core repository

Cool! Thanks, it now "works"...only: I can see the green dot for the activity and the tags are replying...but not appearing on the maps...any help?

[image]

mishailovic, 2023-02-16:
> Cool! Thanks, it now "works"...only: I can see the green dot for the activity and the tags are replying...but not appearing on the maps...any help?
>
> [image]

Can you try again with this build? https://github.com/mishailovic/openhaystack/suites/11027640778/artifacts/559843737

Don't forget to uninstall old mail plugin

Furtivo360, 2023-02-16:
Not working...
- Old mail plugin uninstalled
- downloded your version (Version 5dd4780 (1) - installed
- new mail plug-in installed
Same issue "No reports found"

mishailovic, 2023-02-16:
> Not working...
> 
> * Old mail plugin uninstalled
> * downloded your version (Version [5dd4780](https://github.com/seemoo-lab/openhaystack/commit/5dd478050a8ffe0a10adebc0a5aa246ff630d196) (1) - installed
> * new mail plug-in installed
>   Same issue "No reports found"

Tbh I have no idea how to fix it. I just ordered my esp32 and it will be by the next week. I couldn't test functionality of the app unfortunately. Maybe main developer will take a look at this issue someday.

pcace, 2023-02-17:
> @pcace yeah, this little dot should become green. I think apple silicon is another pain, I'm not really sure what to do with it
i am not really familiar with osx development, but is it possible that this must be build for ARM plattform and currently is build for x86?
Cheers

mishailovic, 2023-02-17:
> do you have your changes somewhere in a different branch/fork?

I did nothing but merging #158 and rewriting build system so I fan trigger it manually

pcace, 2023-02-17:
hm thanks for clarification. 
is there any success story of openhaystack running on ARM?

Thanks a lot!

holygrolli, 2023-03-05:
Thanks for all the input here. I finally got it working on Ventura on my M1 Pro. I had to build it on my own. See my #182 where I also included a separate file for instructions how to build OpenHaystack on your own

marinsss, 2023-05-11:
> Hey guys, little update on this one. I bumped versions of mailplugin and built it using GitHub actions. You can grub it [...]

worked great for me. thank you