seemoo-lab/openhaystack #173: ESP32 is still findable?

beckmx, 2023-01-25:
**Describe the bug**
Seems that esp32 is not findable anymore, I have flashed several devices and none of them seem to notify or at least the "deploy" button stays in there, microbit v1 is findable :)

**To Reproduce**
Steps to reproduce the behavior:
1. Manual compile
2. Flash as in the instructions with the ID
3. The app doesnt allow to upload directly from the app
4. Not able to be found

**Expected behavior**
ESP32 should be able to be found

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**OpenHaystack version:**
0.5.2 (1)

**macOS version:**
Ventura 13.1

jingwei55, 2023-05-09:
Hi may I know if you managed to fix your issue? After flashing the firmware my ESP32 still cannot be found as well.

beckmx, 2023-05-09:
I did managed to do it, most probably you are missing the id of the uuid, it is tricky, if you flash the image as it is, it has to be patch with the uuid

jingwei55, 2023-05-10:
Ah yes that was the case. I keyed in the uuid in the wrong format which was why the firmware could not detect the key. Thanks!