seemoo-lab/openhaystack #175: nRF51822 error: open failed

backupandforkacc, 2023-02-02:
There is this [nice tutorial](https://github.com/seemoo-lab/openhaystack/issues/35) how to transfer the .bin to the nrf51 device, but I fail at step 7. I get the error:

```
openocd -f /usr/local/share/openocd/scripts/interface/stlink-v2.cfg -f /usr/local/share/openocd/scripts/target/nrf51.cfg
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Error: open failed
```
I am unsure how to connect Bat+ and Bat- and which way to put in the battery. I tested several ways here.

backupandforkacc, 2023-02-02:
I also tried in windows after installing openocd via msys64 and got the same error. 
What I did:
1. Download and Install msys64
2. Install openocd in msys64 console (pacman -S mingw-w64-x86_64-openocd)
3. run
 set firmware=C:\Users\simon\Downloads\microbit.bin
then
C:\Users\simon\Downloads\msys64\mingw64\bin\openocd.exe -f C:\Users\simon\Downloads\msys64\mingw64\share\openocd\scripts\interface\stlink.cfg -f C:\Users\simon\Downloads\msys64\mingw64\share\openocd\scripts\target\nrf51.cfg -c init -c "reset halt" -c "nrf51 mass_erase" -c "program %firmware% verify" -c "program %firmware%"

Sn0wfreezeDev, 2023-03-13:
We also added the tutotial to our wiki: https://github.com/seemoo-lab/openhaystack/wiki/Flashing-nRF-with-OpenOCD---ST-Link 

Maybe try following the steps closely