seemoo-lab/openhaystack #177: cheap bluetoothmodule - would it work?

pcace, 2023-02-16:
Hi there, 

would this work for the use with openHaystack? [NRF52832 @aliexpress](https://shrl.pfcace.de/TqGY
)
what would be the smallest yet cheap devices wich would work? 

Thanks a lot!!

mishailovic, 2023-02-17:
currently there is no firmware for that model. In my opinion esp32 is still the best option you can have, and it costs half of the price of the NRF52832 ([link](https://aliexpress.ru/item/32864722159.html?sku_id=12000015991868470))

krakowski, 2023-02-24:
Found these cheap bluetooth beacons using nRF51822 : https://aliexpress.com/item/1005003671695188.html

Using a flasher (ST-LINK/V2) it should be possible to flash the nRF51 firmware (https://github.com/seemoo-lab/openhaystack/tree/main/Firmware/Microbit_v1) onto it. Haven't tried it out yet, though.

jamine16, 2023-02-26:
You can flash them with ST-link v2, but you need to convert it to a J-Link using the Asshole.bin firmware from here first

https://github.com/e0r/stm32f103c8t6-jlink

Then you can use the Jflash software.