seemoo-lab/openhaystack #183: Microbit:V2 Deployment and an Issue.

VincentKaufmann, 2023-03-09:
Im trying to deploy it on a Microbit:v2.

1. The public key is supposed to be 28 digits. I just generated a 28 digit string such as: `MIGeMA0GCSqGSIb3DQEBAQUAA4GM` Is that okay what public/private key should I generate and where should I store the private key or is the public key just a term for a 28 digit identifier in this case?

2. Firmware: I cd into the offline-finding directory and I run make, I get `../blessed/examples/Makefile.common:6: *** PLATFORM is not defined.  Stop.` How do I configure it for the Microbit:v2? I see in the git repo "blessed" and I figured I have to configure blessed for the Microbit:v2 right?

VincentKaufmann, 2023-03-09:
I figured out I was in the wrong directory. From the following directory: Microbit_v1 I can make the firmware but when I try to install it with: `make install DEPLOY_PATH=/Volumes/MICROBIT`.

Here is the build log:

`/Library/Developer/CommandLineTools/usr/bin/make -C blessed
make[1]: Nothing to be done for `all'.
/Library/Developer/CommandLineTools/usr/bin/make -C offline-finding
-e MKDIR         build
-e CC    main.c
-e LD    build/offline-finding.out
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc_nano.a(libc_a-closer.o): in function `_close_r':
closer.c:(.text._close_r+0xc): warning: _close is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc_nano.a(libc_a-lseekr.o): in function `_lseek_r':
lseekr.c:(.text._lseek_r+0x10): warning: _lseek is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc_nano.a(libc_a-readr.o): in function `_read_r':
readr.c:(.text._read_r+0x10): warning: _read is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc_nano.a(libc_a-writer.o): in function `_write_r':
writer.c:(.text._write_r+0x10): warning: _write is not implemented and will always fail
-e SIZE  build/offline-finding.out
   text    data     bss     dec     hex filename
  15580     180     904   16664    4118 build/offline-finding.out
-e OBJCOPY       build/offline-finding.bin`

I get a notification on my MacBook: 
Disk not Ejected Properly. Eject Microbit before disconnecting or turning it off. Inside /Volumes/MICROBIT I got three files.

1. FAIL.TXT:
error: The application image is not compatible with the target.
type: user

2. DETAILS.TXT:
# DAPLink Firmware - see https://daplink.io
Build ID: v0257-gc782a5ba (gcc)
Unique ID: 9906360200052820f9b1ccf9e7bfaa71000000006e052820
HIC ID: 6e052820
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Incompatible image detection: 1
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0257
Bootloader Version: 0257
Git SHA: c782a5ba907377658bc28aa8d132a0fa44543687
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0x725bea7d
Interface CRC: 0xe561f1de
Remount count: 4
URL: https://microbit.org/device/?id=9906&v=0257

3. MICROBIT.HTM

VincentKaufmann, 2023-03-09:
I just figured it out: 

Micro:bit V2 is using the nRF52833 chip. Micro:bit V1 is based on the nRF51822 chip.

In the following directory there is the platform specified: `Microbit_v1/blessed/platform`.

Where can I get the correct blessed platform files for the nRF52833 chip?

koenvervloesem, 2023-03-09:
The blessed repo says:

> Currently, only the [nRF51822] (https://github.com/pauloborges/blessed/tree/devel/platform/nrf51822) SoC from [Nordic Semiconductor](https://www.nordicsemi.com/) is supported.

You can always try my [Zephyr-based OpenHaystack firmware](https://github.com/koenvervloesem/openhaystack-zephyr). The BBC micro:bit v2 is one of the devices I tested it on.

VincentKaufmann, 2023-03-09:
> The blessed repo says:
>
>
> You can always try my [Zephyr-based OpenHaystack firmware](https://github.com/koenvervloesem/openhaystack-zephyr). The BBC micro:bit v2 is one of the devices I tested it on.

I try it out! Will follow your guide to flash the firmware. Is there anything I need to watch out for? It will run with the stock OpenHaystack software right?

Thank you

VincentKaufmann, 2023-03-09:
> If you encounter some issues, please open an issue in my repository, I'm happy to help there.
> 
> > It will run with the stock OpenHaystack software right?
> 
> It works with OpenHaystack's macOS application, yes.

I get this: `zsh: command not found: west`

Just a question, what is west? I see a west.yml file, how do I use it? Is west a pre-requisit?

VincentKaufmann, 2023-03-09:
Could you upload a pre-build firmware for the V2 that I could use to play around with OpenHaystack?

VincentKaufmann, 2023-03-09:
> If you encounter some issues, please open an issue in my repository, I'm happy to help there.
> 
> > It will run with the stock OpenHaystack software right?
> 
> It works with OpenHaystack's macOS application, yes.

Could you upload a pre-build firmware for the V2 that I could use to play around with OpenHaystack?

koenvervloesem, 2023-03-09:
> I get this: `zsh: command not found: west`
> 
> Just a question, what is west? I see a west.yml file, how do I use it? Is west a pre-requisit?

Please install the requirements:

> A [Zephyr development environment](https://docs.zephyrproject.org/latest/getting_started/index.html)

Sn0wfreezeDev, 2023-03-13:
There's also a Microbit_v2 firmware which works. https://github.com/seemoo-lab/openhaystack/pull/84

closed 2023-03-13 by Sn0wfreezeDev