seemoo-lab/openhaystack #184: build/openhaystack.bin NOT THERE

TomOMaley, 2023-03-10:
Hi, after days I managed to run idf.py build. But as I had no clue how to make a new project I used the esp-idf example "hello-word". On this path the files 
build/bootloader/bootloader.bin -- The second stage bootloader
build/partition_table/partition-table.bin -- The partition table
were created but not build/openhaystack.bin.

I'd really appreciate your help

davesenior9, 2023-03-11:
What device are you trying to flash?

TomOMaley, 2023-03-11:
It is an esp32 chip (detail see below). I did the ESPRESSif Standard Toolchain Setup for Linux and macOS with stable Version 5.0.1 using HomeBrew
Mac Version is 12.6.1

The command  screen /dev/cu.usbserial-0001 115200 shows:
Hello world!
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision v1.0, 2MB external flash
Minimum free heap size: 301716 bytes\

TomOMaley, 2023-03-12:
Hi there, my last experience is: in the openhaystack App Package Contents there is the "build/openhaystack.bin".
esp-idf is on usr/esp/idf, whilest openhaystack App is in the download folder. Do I need to bring both folders in one place and if so, how?
When I try to deploy esp32 device from the openhaystack App I get: Could not deploy, Deploying to microbit failed. Please reconnect the device over USB. But reconnection do not help. ????

closed 2023-03-12 by TomOMaley