seemoo-lab/openhaystack #185: Low battery status warning

jrusi, 2023-03-11:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Receive notification of beacon's low battery

**Additional context**
Woud be very cool to see an exclamation mark by the device name informing that the battery is low and should be changed

Sn0wfreezeDev, 2023-03-13:
We are not going to implement this, because it would require us to read the battery from a various differnt types of beacons, which is not possible.

closed 2023-03-13 by Sn0wfreezeDev

bobrippling, 2024-05-28:
Out of interest - from [the paper](https://github.com/seemoo-lab/openhaystack?tab=readme-ov-file#references), the status byte is this one (in the middle):

https://github.com/seemoo-lab/openhaystack/blob/7d72fa1ac19d2a9f6dec43011be07df8976a8b02/Firmware/ESP32/main/openhaystack_main.c#L31-L33

For writing my own firmware, do you know if the representation or encoding of this byte is documented anywhere? I notice that zero appears to represent "full battery"

bobrippling, 2024-09-10:
> For writing my own firmware, do you know if the representation or encoding of this byte is documented anywhere? I notice that zero appears to represent "full battery"

Found this "documented" in [this repo](https://github.com/MatthewKuKanich/FindMyFlipper/blob/e5a678e33dd17af46bb63050be4c9d10e17f6252/FindMyFlipper/findmy_state.c#L126-L137)