seemoo-lab/openhaystack #191: Ventura OS 13.3 - Mailplugin issue

Furtivo360, 2023-03-29:
OS 13.3 got released: new OS, new mail plugin incompatibility.

This is the new UUID for Mail app Version 16.0 (3731.500.231) 

**NEW PluginCompatibilityUUID:** 281F8A5C-0AF9-4BE6-8B8A-C0CB9C2068BE

As always:

1. Disable GateKeeper : sudo spctl --master-disable
2. Add the UUID creating a new string: Supported13.3PluginCompatibilityUUIDs in the Info.plist 
3. codesign --force --deep -s - OpenHaystackMail.mailbundle
4. Re-activate OpenHaystack Mail plugin checkbox in Mail application
5. Re-enabling GateKeeper sudo spctl --master-enable

Done!

Happy days 👍

CoooWeee, 2023-04-04:
@Furtivo360 Thanks, that was what I needed

TylerHallPureEngineering, 2023-04-04:
couple questions, im new and unfamiliar with previous hacks to get this working.

Are you saying to edit the Info.plist to contain:
<key>PluginCompatibilityUUID</key>
<string>281F8A5C-0AF9-4BE6-8B8A-C0CB9C2068BE</string>

or by adding a new key like:
<key> Supported13.3PluginCompatibilityUUIDs</key>
<string>281F8A5C-0AF9-4BE6-8B8A-C0CB9C2068BE</string>

Also, the Info.plist in the mail application says its mounted in a read-only volume. Can give any hint as to how id write to that file?

TylerHallPureEngineering, 2023-04-04:
> couple questions, im new and unfamiliar with previous hacks to get this working.
>
> Are you saying to edit the Info.plist to contain: `<key>PluginCompatibilityUUID</key> [...]

update. I'm learning. my bad. for anyone else new, its the OpenHaystackMail.mailbundle 's Info.plist. There i was able to see the other  SupportedXX.XPluginCompatibilityUUIDs which are of type array, add new row, each can be added as a string. It was stumbled by trying previous issues. https://github.com/seemoo-lab/openhaystack/issues/162

Thanks.

KlausB-S, 2023-04-15:
Sorry, it doesn't work for me.
I added the row in the OpenHaystackMail.mailbundle`s Info.plist and let install the Mailbundle again, but then i receive a error message, that the file is corrupted. 
Could anybody share a working Mailbundle?
Thanks!

nsk, 2023-05-06:
not working for me

HiveTechDev, 2023-05-08:
Here's what's working for me (at the exact moment)

macOS Ventura Version 13.3
Mail Version 16.0 (3731.500.231)

[Info.plist.zip](https://github.com/seemoo-lab/openhaystack/files/11424965/Info.plist.zip)

KlausB-S, 2023-05-09:
still not working... :-(

PrinzEric, 2023-05-16:
some for me, 13.3.1 anyone can take a new info.pls PLEASE ?

PrinzEric, 2023-05-19:
Doesnt work.

DiegoArmstrong, 2023-05-23:
> Here's what's working for me (at the exact moment)
> 
> macOS Ventura Version 13.3 Mail Version 16.0 (3731.500.231) OpenHaystack Version 0.5.2 (1)
> 
> [Info.plist.zip](https://github.com/seemoo-lab/openhaystack/files/11424965/Info.plist.zip)

Tried your file but still not working for me :(

My steps after putting your Info.plist file into the OpenHaystackMail.mailbundle...

1) sudo spctl --master-disable
2) sudo codesign -f -s - ~/Library/Mail/Bundles/OpenHaystackMail.mailbundle
3) Open Mail --> Settings --> General --> Manage Plug-ins --> Enable OpenHaystackMail.mailbundle
4) I receive the same error
"Incompatible Plug-ins Disabled" 
"Mail has disabled the following plug-ins: OpenHaystackMail.mailbundle
Contact the makers of these plug-ins for versions that are compatible with Mail 16.0."

mac OS venture 13.2.1
Mail 16.0 (3731.400.51.1.1)
OpenHaystack Version 0.5.2 

Apologies if I am making a silly error, I am very new to this pls help :)

oz-glenn, 2023-06-01:
I'm on MacOS Ventura 13.4 and this didn't initially work. 
To fix it I copied my recently added plist entry for 13.3 and pasted it replacing 13.3 with 13.4 in the name and it worked.

andaju, 2023-06-08:
That doesn't work for me. When I add the new UID to the 13.4 section [281F8A5C-0AF9-4BE6-8B8A-C0CB9C2068BE], then I always get the message, that „OpenHaystackMail.mailbundle“ ist beschädigt und kann nicht geöffnet werden. Es empfiehlt sich, das Objekt in den Papierkorb zu bewegen.
Same UID in the 13.3 section and I get an "incompatibel Plugin" message.
Any ideas?
mac 13.4 (22F66)
Mail 16.0 (3731.600.7) 
OpenHaystack 0.5.2 (1)

andaju, 2023-06-10:
Now I found it in combining some posts here:
If you modify the info.plist you have to codesign it again with terminal.
This only works, if you give terminal the full access to the disk.
Now OpenHaystack 0.5.2 is working with macOS 13.4 / Mail 16

h1t35h, 2023-08-20:
> Here's what's working for me (at the exact moment)
> 
> macOS Ventura Version 13.3 Mail Version 16.0 (3731.500.231) OpenHaystack Version 0.5.2 (1)
> 
> [Info.plist.zip](https://github.com/seemoo-lab/openhaystack/files/11424965/Info.plist.zip)

Using the above and copying the "Supported13.3PluginCompatibilityUUIDs" setting to "Supported13.4PluginCompatibilityUUIDs". Worked for 13.4.