seemoo-lab/openhaystack #194: Better error messages for flashing on ESP32

scorpi11, 2023-04-15:
**Is your feature request related to a problem? Please describe.**
I tried to flash an ESP32 device. It failed and only got the message that I should reconnect the device via USB.

**Describe the solution you'd like**
I want to see what exactly went wrong, what error message was returned by the shell script and what parameters are provided.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

beckmx, 2023-05-14:
You need to get familiar with flashing directly with esptool, rather than using the option directly on the app

naxiemolv, 2023-05-16:
try para -s
./flash_esp32.sh -p {{your port}} "{{your key}}" **-s**