seemoo-lab/openhaystack #197: Tell me about the `confidence` key

mrmay-dev, 2023-05-08:
Hi all,

Haystack reports have a key called `confidence`. I’ve seen it go as high as 115. What is this number telling me? What’s the scale?

My best guesses are: 

- it’s the dBm of the beacon’s signal. This would allow a very rough estimate of the distance between the tag and the iPhone. 

- it’s the estimated distance, in feet, between the tag and the iPhone sending the report. Using this as a confidence radius produces reasonable results. 

I think the first is most likely, but the second produces reasonable ranges. 

What am I missing or don’t know?

Thanks in advance

michellarcari, 2023-05-19:
I'm also curious to know what that piece of data means. Anyone?

humpataa, 2023-11-08:
I get confidence values up to 220 ... my guess is the distance to the tag: the closer the higher the more accurate the gps values. I believe it is based on the strength of the signal (dBm) which can be used to measure the distance.

humpataa, 2024-01-05:
Just read in the [author's paper](https://github.com/seemoo-lab/openhaystack?tab=readme-ov-file#references) that it likely represents the accuracy in meters. If you open google maps and look at yourself it shows a blue circle – "confidence" being the diameter of it – if I understand it right.

sonman, 2024-12-28:
Just to make it clear:
We have two similar fields in the report.
* confidence (1 byte, not encrypted) (value 0,1,2,3)
* horizontal accuracy (1 byte, encrypted) (e.g. values between 50-255)