seemoo-lab/openhaystack #203: Mail Plug-In

Logy2470, 2023-06-05:
It says "Cannot Connect to the Mail Plug-In. Open Apple Mail and Make sure the plug-in is enabled." and I can't figure out how to change it.

glatteis, 2023-06-12:
This happens on every mail update sadly

Sn0wfreezeDev, 2023-06-21:
The issue is also that from the next Apple macOS Mail Plug-Ins are no longer supported.

glatteis, 2023-06-21:
I'm pretty sure it's possible to use the API directly right? There are multiple examples flying around the issues.

Sn0wfreezeDev, 2023-06-22:
The app accesses the API directly. The issue is that you need the iCloud tokens to do that. You can only generate these tokens over a private Apple API or by using a hack (currently done on Linux) that uses the Apple Music app for Android to generate these tokens.

etalot, 2023-07-23:
> The issue is also that from the next Apple macOS Mail Plug-Ins are no longer supported.

So, is this project died??

I just compiled latest version in Xcode and does't work. I have Mac OS 13.4 and Mail Version 16.0 (3731.600.7)

skorokithakis, 2023-08-15:
@Sn0wfreezeDev Would it be possible to get these keys from the Linux/Android hack and add them to the mobile app directly (and not need the Mac proxy)?

glatteis, 2023-08-15:
> @Sn0wfreezeDev Would it be possible to get these keys from the Linux/Android hack and add them to the mobile app directly (and not need the Mac proxy)?

No. See https://github.com/seemoo-lab/openhaystack/issues/63

skorokithakis, 2023-08-15:
Ah, so there's no way to use the project currently?

glatteis, 2023-08-15:
There is code that can fetch the locations without the app on a Mac. For instance, for our service we have this repo: https://gitlab.com/eselac/openhaybike

skorokithakis, 2023-08-15:
Sorry, I mean even on a Mac. It doesn't seem like Mail will load the plugin for me (it never appears in the list).

skorokithakis, 2023-08-15:
@glatteis is there any documentation for that project? I can't find anything in that repo.

glatteis, 2023-08-15:
> @glatteis is there any documentation for that project? I can't find anything in that repo.

There isn't currently, as we just use that for our own purposes and there was no reason yet to add some (not really intended for direct public use), but I can give you a small walkthrough that should fetch locations for a given tracker:

- install the requirements (preferably through pipenv)
- in the pipenv, run `python scripts/get_icloud_key.py` and paste the output (`icloud_key = ...`) into `config.toml`
- run `python scripts/locations.py ----key-id <your key id> --private-key <your private key>`

this _should_ fetch locations for a tracker - if you have any trouble please send me a message :)

skorokithakis, 2023-08-15:
I'll try that, thank you!

skorokithakis, 2023-08-17:
I can confirm that the OpenHaybike code works great, so presumably it can be implemented into OpenHaystack without needing to rely on the Mail plugin.

23tom2323, 2023-08-23:
I'm currently testing openhaybike.

I have the icloud_key. But I can't get any further.

Where can I find the config.toml where the icloud_key should go in?

And "python scripts/locations.py --key-id <your key id> --private-key <your private key>" is also not clear yet.

Where can I find the <your key id> and <your private key> ?

glatteis, 2023-08-23:
If you write stuff in here it pings all users, please send me questions about the openhaybike repo directly. Just making this more clear:

> I'm currently testing openhaybike.
> 
> I have the icloud_key. But I can't get any further.
> 
> Where can I find the config.toml where the icloud_key should go in?

You create the config.toml yourself, just in the root folder of openhaybike. Or run `cp config.toml.sample config.toml` and paste in your key.

> And "python scripts/locations.py --key-id --private-key " is also not clear yet.
>
> Where can I find the and ?

Your tracker has a key id and a private key - you pass them as arguments to the script. Openhaybike doesn't create the tracker for you, you need to follow openhaystack's instructions for that.

bbsan2k, 2023-08-27:
I think the base of the authentification stuff is a little bit better documented here: https://github.com/Hsn723/MMeTokenDecrypt

In fact it relies on a vulnerability in macOS which is already present since MMe(?) - but if it’s still present it should be pretty straightforward to implement it in openhaystack

Sn0wfreezeDev, 2023-10-09:
I just updated the OpenHaystack code to support the Mail plugin on 13.5 and 13.6

closed 2023-10-09 by Sn0wfreezeDev