seemoo-lab/openhaystack #205: HCI.py usage

DrRob, 2023-06-15:
When I run "sudo python3 HCI.py --key <ADVERTISMENT_KEY>" on a Pi it outputs some debug info and then exits.  Will the Pi continue to broadcast the advertisement after that or does it just broadcast once?  If the latter, how often should the script be run?  If the former, will it continue broadcasting after a reboot or does the script need to be run after each reboot?

Chacha1305Poly20, 2023-10-19:
From reading and execution of the script, the broadcast will send once. To make it sending periodically, modification of the code is required. This isn't too much of work so I do not see necessary for making a feature for now, just loop in python or bash should do the trick.

DrRob, 2023-10-19:
I wasn't sure if the Bluetooth stack on the Pi would repeat the broadcast autonomously.

Anyway, I tried calling the script once per minute, but could never get it to appear in the Find My app or trigger an Unknown tracker alert in my Android phone or be detected by the Apple Tracker Detect app.  So I gave up with it, life's too short.

Harvie, 2024-06-24:
Tried on my thinkpad and it did broadcasted even after the script exited. Running the script continuously might not be that great idea, because the script soft resets whole bluetooth interface (and related stack) each time you run it. That might interfere with broadcasts actualy being sent.

Chapoly1305, 2025-01-24:
We provided an updated hci.py for using on other Linux device than Raspberry. 
https://github.com/Chapoly1305/FindMy?tab=readme-ov-file#advertise-with-improved-hcipy-on-linux