seemoo-lab/openhaystack #211: difficulties with OpenHaystack Version 0.5.2 (1), macOS 13.1 , Mail Version 16.0 (3731.300.101.1.3)

23tom2323, 2023-08-22:
Hello, I have some problems ;-)

When you start it always says "New version available". It also downloads a new file with the same status as I already have.

The next curious thing is that I can't find any plugins in the mail.

Apple Mail → Preferences → General → Manage Plug-Ins... Unfortunately, everything is empty for me

I flashed 3x ESP32 WROOM with the method from https://github.com/mrchi/openhaystack-esp32-firmware

Unfortunately, this was not possible from my OpenHaystack version.

Unfortunately, the ESP32 are not yet displayed in OpenHaystack either. I don't think it works without the mail plugin.

What can I do?

Greetings Tom

Here are my versions:

OpenHaystack Version 0.5.2 (1)  Jun 1, 2022
macOS 13.1 
Mail Version 16.0 (3731.300.101.1.3)

tyhallcsu, 2023-08-23:
https://gist.github.com/tyhallcsu/52a8d058149d395351749d8282314e72

itpilot, 2023-09-10:
Having the same issue but with the following

OpenHaystack Version 0.5.2 (1) Jun 1, 2022
macOS 13.3.1
Mail Version 16.0 (3731.300.101.1.3)

Any chance of a Newly Compiled OpenHaystack that support Mail Version 16?

Any help greatly appreciated