seemoo-lab/openhaystack #214: Mac OS 14 (Sonoma) Support

Loic760, 2023-09-01:
**Issue:**
With the upcoming release (of the currently in beta) macOS 14.0 Sonoma, Apple has announced the removal of mail plugins compatibility. This change will render the OpenHaystack application non-functional.

**Request:**
Find a Workaround to fetch location data of Openhaystack tags that isn't relying on the Apple mail plugin if possible ? (might be with the help of apple iCloud for windows ?) or (minimal hackintosh 13.0 vm without graphical interface as a way to run OHserverapp only as a more efficient way)

Sn0wfreezeDev, 2023-10-09:
I would love to know how AltStore is doing it in macOS Sonoma. If we find out we probably can port it to OpenHaystack. So far we don't have it ready.

rkreutz, 2023-10-14:
@Sn0wfreezeDev would this be an option? https://github.com/seemoo-lab/openhaystack/issues/127

JWegener, 2023-11-23:
As of now, openhaystack is non-functional for those of us who have updated to Sonoma, is that true?

Chapoly1305, 2023-11-30:
> As of now, openhaystack is non-functional for those of us who have updated to Sonoma, is that true?

True. Mail in Sonoma will not show this plug-in. 
[image]

liebrandapps, 2023-12-22:
OpenHayStack seems to be no longer working unless somebody re-implements functionality like the anisette-server [https://github.com/Dadoum/anisette-v3-server](anisette-server). At this point - to my knowledge - there are two alternative options:

[https://github.com/biemster/FindMy][()](https://github.com/biemster/FindMy) 
[https://github.com/liebrandapps/FindMyGUI](https://github.com/liebrandapps/FindMyGUI)

The FindMyGUI (written by me) is based on the work of FindMy by Biemster. I moved this into an app that can be used from a browser. This is still beta. It works for me - however, I did not put any effort into nice appearance. Going to add that.

You will need a Linux system like a Raspberry PI...

fansel, 2024-01-21:
Maybe this project is another option: [macless-haystack](https://github.com/dchristl/macless-haystack/). It's a combination of a couple of popular OpenHaystack implementations.