seemoo-lab/openhaystack #216: project abandoned?

sysadminpower2019, 2023-09-24:
Has this project been abandoned? It seems the last update was over a year ago.

aditya1503, 2023-09-28:
It's still working on Ventura, but I think apple dropped support on Sonoma

MrTalon63, 2023-09-28:
Well, it really depends on way to get back the data from Apple cloud service and as they're only accessible with token that can be created via private API (from what I know?) There need to be a different way rather than relaying on mail plugins. I'm not sure if authors are keen on supporting this software as none of them responded to any recent issues, I think.

MrTalon63, 2023-09-29:
It appears that @Sn0wfreezeDev was the most active last time author.
Please inform us if you're willing to keep up this project or if it is abandoned/legacy and should be forked by anyone who wants to push it forward from now on.

ololobus, 2024-04-05:
> Well, it really depends on way to get back the data from Apple cloud service and as they're only accessible with token that can be created via private API (from what I know?) There need to be a different way rather than relaying on mail plugins.

Just came across an alternative way to access required APIs and posted a proposal as an issue https://github.com/seemoo-lab/openhaystack/issues/240

I've played with it last week and it worked well. I had some security concerns, as every time you run some random code from the internet, which you barely read; but otherwise, no issues at all