seemoo-lab/openhaystack #217: Trying to 'deploy' on a Seeed nRF52840

megamaniac, 2023-10-02:
Hi,

I think perhaps this is a little over my head, but I was hoping maybe you more knowledgeable folks could hold my hand a little trying to deploy OpenHaystack on a Seeed nRF52840.

When hotting 'deploy' I get the following error:

`# Setting up python env in folder /var/folders/qw/25lz8fwx67s1l2wbzgcdg1l80000gn/T/OpenHaystack/venv
# Activate venv and install pynrfjprog and intelhex
Requirement already satisfied: pip in /private/var/folders/qw/25lz8fwx67s1l2wbzgcdg1l80000gn/T/OpenHaystack/venv/lib/python3.9/site-packages (21.2.4)
Collecting pip
Using cached pip-23.2.1-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
Attempting uninstall: pip
Found existing installation: pip 21.2.4
Uninstalling pip-21.2.4:
Successfully uninstalled pip-21.2.4
Successfully installed pip-23.2.1
Collecting pynrfjprog
Using cached pynrfjprog-10.23.0-py3-none-any.whl
Collecting future (from pynrfjprog)
Using cached future-0.18.3-py3-none-any.whl
Collecting tomli-w (from pynrfjprog)
Using cached tomli_w-1.0.0-py3-none-any.whl (6.0 kB)
Installing collected packages: tomli-w, future, pynrfjprog
Successfully installed future-0.18.3 pynrfjprog-10.23.0 tomli-w-1.0.0
Collecting intelhex
Using cached intelhex-2.3.0-py2.py3-none-any.whl (50 kB)
Installing collected packages: intelhex
Successfully installed intelhex-2.3.0
### Executing python script ###
[*] Opening API with device family UNKNOWN, reading the device family.
Traceback (most recent call last):
File "/var/folders/qw/25lz8fwx67s1l2wbzgcdg1l80000gn/T/OpenHaystack/flash_nrf.py", line 120, in <module>
flash_openhaystack_fw(public_key=b64decode(args['public_key']), symmetric_key=b64decode(args['symmetric_key']), update_interval=args['update_interval'], hex_path=args['path_to_hex'])
File "/var/folders/qw/25lz8fwx67s1l2wbzgcdg1l80000gn/T/OpenHaystack/flash_nrf.py", line 31, in flash_openhaystack_fw
with LowLevel.API(
File "/private/var/folders/qw/25lz8fwx67s1l2wbzgcdg1l80000gn/T/OpenHaystack/venv/lib/python3.9/site-packages/pynrfjprog/LowLevel.py", line 2472, in __enter__
self.open()
File "/private/var/folders/qw/25lz8fwx67s1l2wbzgcdg1l80000gn/T/OpenHaystack/venv/lib/python3.9/site-packages/pynrfjprog/LowLevel.py", line 258, in open
raise APIError(result, error_data=self.get_errors())
pynrfjprog.APIError.APIError: An error was reported by NRFJPROG DLL: -101 JLINKARM_DLL_COULD_NOT_BE_OPENED.
### done
`

I get this error wether my Seeed device is plugged in or not. Just wondering if I have missed something?

koenvervloesem, 2023-10-02:
See my answer to your question here: https://github.com/seemoo-lab/openhaystack/discussions/212

megamaniac, 2023-10-04:
Apologies, I completely missed that that was Openhaystack-Zephyr, rather than the original. 

I shall give that a go.