seemoo-lab/openhaystack #219: iPhone 15 - lack of reports

davesenior9, 2023-10-28:
I've recently upgraded to an iPhone 15 and I am noticing a distinct lack of reports, generated by that device.

My old phone (iPhone 8), would routinely upload reports, and I would subsequently see a location report of my home.

Since getting the iPhone 15, I have got ZERO reports from my OpenHayStack devices. I haven't changed the beacon device, if I leave it in my bag and go to work, I'll get reports as per usual from the wider Apple Offline Finding network.

Does anyone have any insight's on how the iPhone 15 might be doing this? Or has anyone seen similar behavior?

shiprec, 2023-12-06:
@davesenior9 Where you able to get this resolved?

davesenior9, 2023-12-19:
Unfortunately no. I tried changing the status field to 'imitate' other Apple Devices, such as Airpods an actual Airtag etc, didn't see any change in result.

shiprec, 2023-12-19:
Not sure if you saw this: 
https://github.com/biemster/FindMy/issues/40

biemster, 2023-12-24:
@davesenior9 what exactly did you do to the status field?

Systm21, 2023-12-29:
@davesenior9 the guys at https://github.com/biemster/FindMy/issues/40 already fiddled around with Status Bytes and Hint Bytes, but without any success. 
If you have logs from apple devices and maybe the adv data, it can help, finding a solution.

isibizi, 2024-03-22:
iOS 17.4.1 has fixed this issue

davesenior9, 2024-03-22:
So the Off-line Finding network is working as it was with IOS 16?

isibizi, 2024-03-22:
Yes, is definitely fixed. I have an 15 pro max with 17.4.1

davesenior9, 2024-03-22:
That's huge news. I'll give it a look/try this morning.