seemoo-lab/openhaystack #232: Q: Does anyone know how to debug FindMy?

th122, 2024-02-03:
This is more a question related to the original FindMy, but I'm desperate. I had borrowed an ipad to register a couple of airtag"clones", For viewing an old MacMini was updated to 10.12 and 10.13 with the help of OpenCore. The Tags went into items that are at risk to attract theft (A tractor, some field equipment parked outside, a camper, a car-trailer for transporting the tractor), and the ipad was returned to its original owner (University). Things went smooth until after a few weeks, the positions stopped updating. 
Yesterday I discovered that the car-trailer had disappeared, leaving only the remains of one of the locks curb-side. And the last known position FindMy has was from December.

I re-borrowed the iPad, and sure, it does show all my devices, but no luck with any items. I also cannot register a new tag (Server unavailable), Or ask information about one of the other tags when scanned with that iPad.

Does anyone know how to debug FindMy? 
Any ideas? 
I'll also give the Haystack a try, hoping it can retrieve more recent data on the tags.

Thanks in advance

Itheras, 2024-02-05:
I can help you. what information do you have available. do you have the keys ?

th122, 2024-02-06:
Thanks! -I still have them listed on one desktop (running ventura) but the  most recent location is from Dec24, and threre are no updates on any of the tags. They seem to have been lost from the iCloud keychain. 
` $HOME/Library/Caches/com.apple.findmy.fmipcore/Items.data` seems to be the most detailled data I can still access.  Would the identifier listed there correspond to the public key at least?

Itheras, 2024-02-06:
If you have the base64 private keys i spinup a page you could test to see if you get updated locations. Let me know if it works for you.

http://97.104.49.146:8075

th122, 2024-02-06:
Found the encrypted keyfiles at `Library/com.apple.icloud.searchpartyd/BeaconProductInfoRecord/` (Ventura) , and converted to xml. Now trying if I'm able to extract them. Does your tool really need (only) the private key? I was under the impression that the public key would be used as identifier to select the payload, and the private key for decryption?
(I have no qualms to publish that private key if I'm able to retrieve it - if it leads to the recovery of the trailer, that's a tag well spent, and if it doesn't, it's lost anyway)

I'm still puzzling what killed the Keychain (if that's what happened). and why the tags went frozen. all files in `Library/com.apple.icloud.searchpartyd/BeaconProductInfoRecord/` have an identical timestamp from Dec24. (I had been doing some installation work on another Macbook that day, and indeed, on that MacBook, the Keychain does behave "weird" details left out to avoid distressing)

Itheras, 2024-02-06:
Yes just the private key i use this to test for me personally. so the script on the back just derives all the keys from the private b64 key then fetch, decrypt and put it on the map on the spot.

Itheras, 2024-02-06:
If you prefer i could modify it how ever you need. I could modify to just fetch the data encrypted so you can decrypt yourself. How ever you need anything let me know.

th122, 2024-02-06:
Now that's a bit deflating. I was hoping for a key pair, but after decrypting the file using the swift script referenced in #37, the file only contained a bunch of metadata that doesn't even look worth encrypting (the identifier, as already part of the filename, some icon references, battery level and manufacturer, and a garbled version string.).
Attached that other MacBook in target mode, and it doesn't have the keys anywhere either. There's only one System I'n looking at right now with a MasterBeacons subdirectory, and that is containing only two files. decrypting those, to take a look, but they're definitely not the Tag I'm looking for. (Still curious what they might be) 
Would the Identifier of the Tag be of any use in getting at least encrypted location data, and a possible moving profile, even if the positions are as of yet unknown?

th122, 2024-02-06:
Yesss! The Monterey boot partition hit gold. The directory is called `OwnedBeacons` there, the files decrypt with the same BeaconStore key I already had retrieved. **Got Keys!**
Could you please, please, reactivate your test page?

Itheras, 2024-02-06:
> Yesss! The Monterey boot partition hit gold. The directory is called `OwnedBeacons` there, the files decrypt with the same BeaconStore key I already had retrieved. **Got Keys!**
> 
> Could you please, please, reactivate your test page? 

Is back up 😄

Itheras, 2024-02-06:
Wait have you fix something really quick

Itheras, 2024-02-06:
Sorry i was doing some unrelated changes it should be working now for you.

th122, 2024-02-06:
Thanks - doesn't look like it gets a fix. Was wondering if it wouldn't need the public key as well to identify the dataset to apply the private key to. Of course the tag may have gotten detected and disabled. I'll exhume another key pair, of which I still have the tag and know that it should be in the cache (been walking around with it on campus recently/last Friday)

th122, 2024-02-06:
hm, the key I tried was from he testate, that went around last Friday. So it should have given a reading. I did rejoin the lines from the output after decrypting. I'll paste both keys in succession, once private, and once public - feel free to capture them.

th122, 2024-02-07:
I am still searching how I could stuff those keys into OH (up and running) or one of the developments building upon it), to try to get them back into tracking, even if my account seems to be unable to do so. Another attempt worth pursuing might be to migrate the user account fron the Monterey boot disk, to see if it takes the items along in a functioning environment.

Itheras, 2024-02-07:
Looking at the logs i can see one key returned results but the rest did not. I can only see the public key i dont log the private one.

Itheras, 2024-02-07:
Wait is this a regular airtag? Or a OHS tag? 😅

Itheras, 2024-02-07:
If this is a regular airtag we need to approach this a completely different way.

th122, 2024-02-07:
it's a regular AirTag-clone (one of those without the precision) The manufacturer identifies itself as "Supra GmbH", the Model is called "Maginon Smart Tag".

Itheras, 2024-02-07:
Aaa i understand now sorry. Ok let me rethink this. You have the most important part that is the private key. I will modify my script tomorrow and let you know when is ready to test.

th122, 2024-02-07:
Thank you so much! I´ll get some sleep as well.

jelle619, 2024-04-01:
Okay, I think I see where the problem is, although I do not know how to solve this. My reasoning might not be entirely correct, but I hope it's at least in the right direction.

It seems that tags created with OH always emit the same public key. However, Find My-certified AirTags also have a so-called "shared key" which is used to create a rolling public key. OH simply derrives the public key from the private key, but the way a public key is derrived from a Find My-tag is a little more complicated, involving a combination of this shared key, the private key as some sort of seed (perhaps the date).

I do not know if it possible to supply the secondary key to OH, but I am relatively certain that OH can't be querying the correct public key without knowledge of the secondary key.

etfriedman, 2024-04-04:
Hi all. I'm currently working on an advertisement key generation script based on section 6.1 of "[Who Can Find My Devices?](https://arxiv.org/abs/2103.02282)", taking the decrypted keys from the plist files of a newly paired iPhone. I'm unable to match any of the generated keys with the ones being broadcast by the device. 

@Itheras it seems you might have figured this out? If so I'd appreciate any guidance you can provide on an implementation. I noticed https://github.com/positive-security/find-you managed to get this working, but they helpfully left out the script for the reader to do themselves... Thanks!

wilkyconsultants, 2024-10-17:
Hi folks, wondering if any further success on this initiative as I want to decrypt the files for airtag locations but after Sonoma 14.4 updates the files in ~/Library/Caches/com.apple.findmy.fmipcore/ Items.data and Devices.data are encrypted. Before they were in the clear so it was easy to harvest the location data on a mac, now seems impossible without getting the keys to decrypt the files.

henrik242, 2025-09-05:
@wilkyconsultants There's a decryptor in [findmy-cache-decryptor](https://github.com/Pnut-GGG/findmy-cache-decryptor), but I haven't tried it myself.