seemoo-lab/openhaystack #234: NRF firmware source code

rkreutz, 2024-03-04:
Hey hello,

I've noticed the macOS app has some hex files for flashing an NRF chip with a firmware that supports rolling keys. I couldn't find any source code to those compiled hex files, does anyone know where to find them? I need to do some small changes on ports/pins in it to get it running on some cheap beacons I have.

Thanks

dakhnod, 2024-05-22:
you can check out github.com/dakhnod/faketag as a starting point. It only supports a static list of keys, though.

dienvx1997bn, 2025-02-07:
I have tested with this project and succeeded
[https://github.com/pix/heystack-nrf5x/tree/master](url)

The above project version uses SDK nRF5_SDK_12.3.0_d7731ad. 
If you want to debug, you can use a sample project (e.g ble_app_beacon) on KeilC, then import main.h, main.c and ble_stack.h, ble_stack.c.
Here is the config for my beacon, using nrf51822 chip on keilC. It worked!
[image]

dakhnod, 2025-02-07:
That would be way cooler if it was FCC/make based, as all projects in the nrf ask are!