seemoo-lab/openhaystack #240: Avoid using Mail.app plugin for accessing required APIs

ololobus, 2024-04-05:
Disclaimer: I didn't know about `openhaystack` project and came here with a strange trajectory, so sorry for my ignorance if I'm talking nonsense. Yet, the problem is the same -- use Apple item locations network to track non-Apple certified devices, so here is the idea.

In the README you say

> The OpenHaystack application requires a custom plugin for Apple Mail. It is used to download location reports from Apple's servers via a private API (technical explanation: the plugin inherits Apple Mail's entitlements required to use this API). Therefore, the installation procedure is slightly different and requires you to temporarily disable [Gatekeeper](https://support.apple.com/guide/security/gatekeeper-and-runtime-protection-sec5599b66df/1/web/1).

However, as I mentioned I came here from this repository: https://github.com/MatthewKuKanich/FindMyFlipper/tree/main/AirTagGeneration

It does pretty much the same as your project, it's just not that polished. So what I did:
1. generate a key pair
2. used app on my Flipper Zero to broadcast BLE beacons using this key, when I went for a walk
3. registered a blank Apple ID
4. used scripts in the aforementioned repo to get locations history for the past few hours and it worked flawlessly

I've spent a couple of hours reading the code, and from what I've got, in 4. it doe following:
1. Uses this project https://github.com/Dadoum/anisette-v3-server to generate some random, but specific device IDs
2. Uses Apple ID credentials to auth in icloud
3. Then uses `https://gateway.icloud.com/acsnservice/fetch` with auth info to finally get coordinates

I hope I didn't miss something, while reading the sources and didn't leak any sensitive info, when I was experimenting with it :) But it looks like it just works and doesn't require any complicated Mail.app patching.

What do you think about adopting the same approach `openhaystack`? It'd simplify installation and potentially allow it's usage on non-macOS devices (not sure).

darthnithin, 2024-04-10:
I think https://github.com/dchristl/macless-haystack uses this method. That's why its not reliant on Mac hardware or a Mail plugin

darthnithin, 2024-04-10:
So does https://github.com/biemster/FindMy

ololobus, 2024-04-13:
Thanks for the links!