seemoo-lab/openhaystack #245: Support for Android / Google's Find my Device?

JsBergbau, 2024-06-02:
Google recently launched its find my device network. https://blog.google/products/android/android-find-my-device/

Contrary to Apple, the specification is publicly available https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn 

Is there planned to support Googles Network as well?

denysvitali, 2024-06-12:
The new API seems to be called "spot":

```
izm.h("canary-spot-pa.sandbox.googleapis.com");
izm.h("dev-spot-pa.sandbox.googleapis.com");
izm.h("spot-pa.googleapis.com");
izm.h("spot-pa.googleapis.com");
```

This (protobuf?) API requires the following scopes:

```
https://www.googleapis.com/auth/spot
https://www.googleapis.com/auth/android_device_manager)
```

It looks like it contains the following RPC services:

```
    izt.d("CreateUser", kcl17);
    izt.d("GetUser", kcl12);
    izt.d("UpdateUser", kcl10);
    izt.d("DeleteUser", kcl6);
    izt.d("CreateBleDevice", kcl5);
    izt.d("UpdateBleDevice", kcl2);
    izt.d("UpdateBleDeviceSharingSettings", kcl4);
    izt.d("ListBleDevices", kcl7);
    izt.d("ListEidsForBleDevices", kcl14);
    izt.d("GetEidInfoForE2eeDevices", kcl3);
    izt.d("UploadPrecomputedPublicKeyIds", kcl8);
    izt.d("RotateBleDeviceKeys", kcl13);
    izt.d("GetBleDevice", kcl9);
    izt.d("GetScannedDeviceInfo", kcl16);
    izt.d("DeleteBleDevice", kcl15);
    izt.d("GetServicePublicKey", kcl11);
    izt.d("UploadClientActionResult", kcl1);
```

adriancampos, 2024-06-13:
The FMD app also makes some POST requests to
```
https://android.googleapis.com/nova/nbe_execute_action
```
and
```
https://android.googleapis.com/nova/nbe_list_devices
```

With the proper Authorization Bearer token, I get a list of my device names and images (including "spot" devices) along with a bunch of other binary data in the `application/octet-stream` response.

leonboe1, 2024-08-27:
> The FMD app also makes some POST requests to
>
> ```
> https://android.googleapis.com/nova/nbe_execute_action
> ```
>
> and
>
> ```
> https://android.googleapis.com/nova/nbe_list_devices
> ```
>
> With the proper [...]

Did you figure out if acquiring the Bearer token could be automated?

biemster, 2024-09-25:
@denysvitali or @adriancampos did you find out more on the API side? I implemented the FMD advertisement, where the Eid (ephemeral id) seems the key to all this. Next step would be to retrieve reports with an Eid, which might require registering it first on your google account.

dylanmazurek, 2024-11-04:
I started building out an sdk for the findmy api.

https://github.com/dylanmazurek/google-findmy

denysvitali, 2024-12-29:
It looks like someone is doing that in Python too:
https://github.com/leonboe1/GoogleFindMyTools/

dkgs2000, 2025-03-28:
The  problem seems to be solved! At least the advertisement packet and a working location cycle. Nordic even released a fully functioning example:
https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/app_dev/device_guides/nrf52/index.html#ug-nrf52

Has anyone tried this?

leonboe1, 2025-03-28:
Possibly most of you have found my repository, it was already posted above. But just to let you know, it allows you to use ESP32 trackers with the Google Find My Device network, just like OpenHaystack.

https://github.com/leonboe1/GoogleFindMyTools/

dkgs2000, 2025-03-29:
> Possibly most of you have found my repository, it was already posted above. But just to let you know, it allows you to use ESP32 trackers with the Google Find My Device network, just like OpenHaystack.
> 
> https://github.com/leonboe1/GoogleFindMyTools/

Well yes, but the one i posted is for nRF boards and also supports fast pair. There is no links between the two.

leonboe1, 2025-03-29:
Yes if you want to not use any additional software, only your phone then a Fast Pair-compatible tracker makes more sense. However, I'm not sure if the firmware listed above will work, since Fast Pair normally needs a pre-shared key which is privately stored on the Fast Pair device, and I doubt that this key is included in the firmware. But maybe it is 🤷🏻