seemoo-lab/openhaystack #265: openhaystack as python service to detect the beacons

AnilKumarKandikonda, 2025-02-21:
Hi, 

can you guys help me how to create a python service for openhaystack , i could able to fetch the dsid and auth token and could abel to generate auth headeder, but i am getting 401 error and this is my sample code which i am using right now, any help on this is hihgly apreciated and i am running this python code in mac os 11.5 only.

openhaystack application is running, mail is opened and works fine, but i want to integrate new verison with my own service with openhaystack.
does openhaystack provides any package or how this can be done

import requests
import base64
import datetime

# --- AUTH HEADER ---
dsid = "*************"
auth_token = "*************************************"
credentials = f"{dsid}:{auth_token}"
auth_header = base64.b64encode(credentials.encode()).decode()
print("Authorization Header:", auth_header)

# --- HEADERS ---
headers = {
    "Authorization": f"Basic {auth_header}",
    "User-Agent": "com.apple.iCloudHelper/1.0 (Windows NT 10.0; Win64; x64)",
    "Content-Type": "application/json",
    "X-Apple-I-Client-Time": datetime.datetime.utcnow().isoformat() + "Z",
    "X-Apple-I-TimeZone": "UTC",
    "X-Apple-Locale": "en_US",
}
print("Headers:", headers)

# --- PUBLIC KEY CONVERSION ---
uncompressed = "BN3VU7limfjEMojbbnWqqz8PkquJIM22/EOOpegfVUyZlMgk+2VHfPvvzUd4rJFHYpkMA8x5egjr"
pubkey_bytes = base64.b64decode(uncompressed)
y_parity = pubkey_bytes[-1] % 2
compressed = bytes([0x02 + y_parity]) + pubkey_bytes[1:33]
data_field = base64.b64encode(compressed).decode()
print("Data Field:", data_field)

# --- PAYLOAD ---
payload = {
    "search": [
        {
            "id": "R2Ltvw5IkGKi3kw4RU5fr8tAayCMip/V71Q0on6YZcw=",
            "data": data_field,
            "previous": ""
        }
    ]
}
print("Payload:", payload)

# --- REQUEST ---
url = "https://gateway.icloud.com/acsnservice/fetch"
response = requests.post(
    url,
    json=payload,
    headers=headers
)

print(f"Status: {response.status_code}")
print(response.text)

**security find-generic-password -s iCloud -g**
i am using this to generate the dsid and auth token
i dont  know whether this is right or wrong

i am getting 401 error all the time.

please help me guys to solve this issue

Systm21, 2025-02-21:
https://github.com/dchristl/macless-haystack

From this Git, you can use what you need.

AnilKumarKandikonda, 2025-02-21:
Thank you i will give a try

AnilKumarKandikonda, 2025-02-21:
@Systm21 Thank you, with the link you shared, I could able to achieve the results needed.
Thank you very much

closed 2025-02-24 by AnilKumarKandikonda